fetchrobotics/fetch_gazebo #15: Release for Kinetic

mikeferguson, 2016-08-23:
Dependent repositories still to be released into Jade:
 * [ ] robot_controllers
 * [ ] robot_calibration (upstream blocker: https://github.com/mikeferguson/robot_calibration/issues/29)
 * [ ] fetch_ros (fetch_description, fetch_navigation, fetch_moveit_config)
 * [ ] simple_grasping
 * [ ] moveit_python (upstream blocker: https://github.com/mikeferguson/moveit_python/issues/8)
 * [ ] moveit

Fixes to fetch_gazebo (gazebo7 branch)
 * [x] update package.xml depends for gazebo 7

Possible improvements:
 * [ ] use new friction models to improve simulation

mikeferguson, 2016-08-23:
@davetcoleman -- i've now driven a fetch around in simulation on 16.04 -- I'll continue with actually moving the arm with moveit in the morning -- the repos listed above are what I've had to check out in source (and I have a pretty severe hack in moveit_python to handle the upstream mess that is pyassimp in 16.04)

mikeferguson, 2016-08-23:
My current setup:

```
cd <catkin_workspace>
cd src
git clone https://github.com/fetchrobotics/robot_controllers.git
git clone https://github.com/fetchrobotics/fetch_gazebo.git -b gazebo7
git clone https://github.com/fetchrobotics/fetch_ros.git -b kinetic_fixes
touch fetch_ros/fetch_calibration/CATKIN_IGNORE
git clone https://github.com/mikeferguson/moveit_python.git -b pyassimp_fix
git clone https://github.com/mikeferguson/simple_grasping.git 
git clone https://github.com/ros-perception/open_karto.git
git clone https://github.com/ros-perception/slam_karto.git
git clone https://github.com/ros-perception/sparse_bundle_adjustment.git
git clone https://github.com/ros-planning/moveit.git
cd <catkin_workspace>
rosdep install --from-paths src --ignore-src --rosdistro kinetic -y
catkin_make
```

Now blocking on broken build of moveit

mikeferguson, 2016-08-23:
I have now successfully run the demo under 16.04

mikeferguson, 2016-08-23:
If you don't have desktop-full, you can end up with the laser not working, so we are probably missing a depend

mikeferguson, 2016-12-27:
gazebo7 branch has been rebased on top of gazebo2/5 branches, so it can be released as 0.9.x

zenetio, 2017-07-15:
I used the script suggested by @mikeferguson and after finish install, using roslaunch stockroom_bot stockroomlaunch, I got no launch error and could get the complete scenario with world, bins and the simple arm robot. Then after couple minutes, I got a "not enough memory" message and got an error in terminal window. So, I closed the session and started a new one. Then I started to get an error regarding to tf and the scenario does not show the robot any more. Then I tried other times and did continue getting the same error and no more robot model in the scenario. So, only in the first time I could get the complete scene. Any hints on this?

mikeferguson, 2018-02-13:
FYI, robot_calibration is now in Kinetic. Also, unassigning myself from this.

derkbreeze, 2018-10-12:
Hello,

I'm using Kinetic in Ubuntu 16.04. I followed the aforementioend installation steps. That said, when I tried to catkin build the workspace, I got the follow tf error and I didn't find a solution. Please let me know if you have any suggestion. Thanks.

Errors     << fetch_depth_layer:make /home/lishuai/logs/fetch_depth_layer/build.make.001.log                                                                                                                                                                                      
/home/lishuai/src/fetch_ros/fetch_depth_layer/src/depth_layer.cpp: In member function ‘virtual void costmap_2d::FetchDepthLayer::onInitialize()’:
/home/lishuai/src/fetch_ros/fetch_depth_layer/src/depth_layer.cpp:148:106: error: no matching function for call to ‘tf2_ros::MessageFilter<sensor_msgs::Image_<std::allocator<void> > >::MessageFilter(message_filters::Subscriber<sensor_msgs::Image_<std::allocator<void> > >&, tf::TransformListener&, std::__cxx11::string&, int, ros::NodeHandle&)’
     new tf2_ros::MessageFilter<sensor_msgs::Image>(*depth_image_sub_, *tf_, global_frame_, 10, private_nh));
                                                                                                          ^
In file included from /home/lishuai/src/fetch_ros/fetch_depth_layer/include/fetch_depth_layer/depth_layer.h:40:0,
                 from /home/lishuai/src/fetch_ros/fetch_depth_layer/src/depth_layer.cpp:32:
/opt/ros/kinetic/include/tf2_ros/message_filter.h:190:3: note: candidate: template<class F> tf2_ros::MessageFilter<M>::MessageFilter(F&, tf2::BufferCore&, const string&, uint32_t, ros::CallbackQueueInterface*)
   MessageFilter(F& f, tf2::BufferCore& bc, const std::string& target_frame, uint32_t queue_size, ros::CallbackQueueInterface* cbqueue)
   ^
/opt/ros/kinetic/include/tf2_ros/message_filter.h:190:3: note:   template argument deduction/substitution failed:
/home/lishuai/src/fetch_ros/fetch_depth_layer/src/depth_layer.cpp:148:106: note:   cannot convert ‘*((costmap_2d::FetchDepthLayer*)this)->costmap_2d::FetchDepthLayer::<anonymous>.costmap_2d::VoxelLayer::<anonymous>.costmap_2d::ObstacleLayer::<anonymous>.costmap_2d::CostmapLayer::<anonymous>.costmap_2d::Layer::tf_’ (type ‘tf::TransformListener’) to type ‘tf2::BufferCore&’
     new tf2_ros::MessageFilter<sensor_msgs::Image>(*depth_image_sub_, *tf_, global_frame_, 10, private_nh));
                                                                                                          ^
In file included from /home/lishuai/src/fetch_ros/fetch_depth_layer/include/fetch_depth_layer/depth_layer.h:40:0,
                 from /home/lishuai/src/fetch_ros/fetch_depth_layer/src/depth_layer.cpp:32:
/opt/ros/kinetic/include/tf2_ros/message_filter.h:168:3: note: candidate: tf2_ros::MessageFilter<M>::MessageFilter(tf2::BufferCore&, const string&, uint32_t, ros::CallbackQueueInterface*) [with M = sensor_msgs::Image_<std::allocator<void> >; std::__cxx11::string = std::__cxx11::basic_string<char>; uint32_t = unsigned int]
   MessageFilter(tf2::BufferCore& bc, const std::string& target_frame, uint32_t queue_size, ros::CallbackQueueInterface* cbqueue)
   ^
/opt/ros/kinetic/include/tf2_ros/message_filter.h:168:3: note:   candidate expects 4 arguments, 5 provided
/opt/ros/kinetic/include/tf2_ros/message_filter.h:146:3: note: candidate: template<class F> tf2_ros::MessageFilter<M>::MessageFilter(F&, tf2::BufferCore&, const string&, uint32_t, const ros::NodeHandle&)
   MessageFilter(F& f, tf2::BufferCore& bc, const std::string& target_frame, uint32_t queue_size, const ros::NodeHandle& nh)
   ^
/opt/ros/kinetic/include/tf2_ros/message_filter.h:146:3: note:   template argument deduction/substitution failed:
/home/lishuai/src/fetch_ros/fetch_depth_layer/src/depth_layer.cpp:148:106: note:   cannot convert ‘*((costmap_2d::FetchDepthLayer*)this)->costmap_2d::FetchDepthLayer::<anonymous>.costmap_2d::VoxelLayer::<anonymous>.costmap_2d::ObstacleLayer::<anonymous>.costmap_2d::CostmapLayer::<anonymous>.costmap_2d::Layer::tf_’ (type ‘tf::TransformListener’) to type ‘tf2::BufferCore&’
     new tf2_ros::MessageFilter<sensor_msgs::Image>(*depth_image_sub_, *tf_, global_frame_, 10, private_nh));
                                                                                                          ^
In file included from /home/lishuai/src/fetch_ros/fetch_depth_layer/include/fetch_depth_layer/depth_layer.h:40:0,
                 from /home/lishuai/src/fetch_ros/fetch_depth_layer/src/depth_layer.cpp:32:
/opt/ros/kinetic/include/tf2_ros/message_filter.h:126:3: note: candidate: tf2_ros::MessageFilter<M>::MessageFilter(tf2::BufferCore&, const string&, uint32_t, const ros::NodeHandle&) [with M = sensor_msgs::Image_<std::allocator<void> >; std::__cxx11::string = std::__cxx11::basic_string<char>; uint32_t = unsigned int]
   MessageFilter(tf2::BufferCore& bc, const std::string& target_frame, uint32_t queue_size, const ros::NodeHandle& nh)
   ^
/opt/ros/kinetic/include/tf2_ros/message_filter.h:126:3: note:   candidate expects 4 arguments, 5 provided
/opt/ros/kinetic/include/tf2_ros/message_filter.h:104:7: note: candidate: tf2_ros::MessageFilter<sensor_msgs::Image_<std::allocator<void> > >::MessageFilter(tf2_ros::MessageFilter<sensor_msgs::Image_<std::allocator<void> > >&)
 class MessageFilter : public MessageFilterBase, public message_filters::SimpleFilter<M>
       ^
[... cut ...]
/home/lishuai/src/fetch_ros/fetch_depth_layer/src/depth_layer.cpp:285:10: error: ‘class tf::TransformListener’ has no member named ‘transform’
     tf_->transform(vector, vector, msg->header.frame_id);
          ^
/home/lishuai/src/fetch_ros/fetch_depth_layer/src/depth_layer.cpp:293:92: error: no matching function for call to ‘tf::TransformListener::lookupTransform(const char [10], const _frame_id_type&, const _stamp_type&)’
       transform = tf_->lookupTransform("base_link", msg->header.frame_id, msg->header.stamp);
                                                                                            ^

varunagrawal, 2019-01-25:
@mikeferguson any updates on this? fetchrobotics/fetch_ros#52 is already merged and it would be nice to have the gazebo simulation working with Kinetic.

moriarty, 2019-01-25:
Because libgazebo9-dev is for 18.04 and Melodic. If you want to try on Kinect you’ll need to checkout the gazebo7 branch.

We’re currently preparing to release Melodic ASAP with the simulation environment for “FetchIt!” https://discourse.ros.org/t/announcement-win-a-robot-fetchit-the-fetch-mobile-manipulation-challenge-icra-2019-competition/7569

Currently if you’re on 18.04 and Melodic, from GitHub most of the packages should work.

Not all are available as debians, and it is our firmware/drivers which have been holding us back (they’re not public- just distributed as binaries) as a lot has changed and we want to have a smooth process for customers who have physical robots. Our target is the end of this month.

But if you’re on 18.04 with Melodic, or checkout the gazebo7 branch most things should work.

varunagrawal, 2019-01-25:
gazebo7 won't work because gazebo doesn't provide the ROS wrapper packages for Kinetic. gazebo8 won't work because to get `moveit` to compile, we need the melodic branch as detailed on ros-planning/moveit_tutorials#262.

At this point, your only two releases are either out of LTS or not yet ready to be released. :confused:

moriarty, 2019-01-26:
Are you only trying to build the simulation? Or also run on a real robot?

14.04 & Indigo is still LTS. Until April 2019.

We aren’t targeting Kinetic.

Melodic will be released soon, but will work from source already.

For the physical robots, you’ll need to wait a little longer for the drivers/firmware and instructions.

That MoveIt! Tutorial you’ve linked to, seems to mention that building it from source will work. If you’re not working on a real robot, then building from source should work. If you’re trying to get this on a real robot, then reach out via our support.

moriarty, 2019-01-27:
@varunagrawal I'm going to close this issue and open new one for Melodic to avoid confusion.

```
sudo apt install ros-melodic-ros-base
mkdir -p ~/ros/melodic/fetch-gazebo/src
cd ~/ros/melodic/fetch-gazebo
catkin init
catkin config --extend /opt/ros/melodic
cd src
git clone git@github.com:fetchrobotics/fetch_gazebo
git clone git@github.com:fetchrobotics/fetch_ros
cd ..
rosdep install --from-paths src --ignore-src --rosdistro melodic -y
catkin build
```

```
test_18_04@test-18:~/ros/melodic/fetch-gazebo/src$ wstool info 
workspace: /home/test_18_04/ros/melodic/fetch-gazebo/src

 Localname    S SCM Version (Spec)     UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------    - --- --------------     -----------  ---------------------------
 fetch_ros      git melodic-devel  (-) 4c1a5186f477 git@github.com:fetchrobotics/fetch_ros
 fetch_gazebo   git gazebo9  (-)       fce68bf44ce2 git@github.com:fetchrobotics/fetch_gazebo
```

```
Finished  <<< fetch_gazebo_demo                  [ 1.5 seconds ]                                                                                                                                               
[build] Summary: All 10 packages succeeded!                                                                                                                                                                    
[build]   Ignored:   None.                                                                                                                                                                                     
[build]   Warnings:  2 packages succeeded with warnings.                                                                                                                                                       
[build]   Abandoned: None.                                                                                                                                                                                     
[build]   Failed:    None.                                                                                                                                                                                     
[build] Runtime: 12.2 seconds total.                                                                                                                                                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.
test_18_04@test-18:~/ros/melodic/fetch-gazebo/src$ 
```

There are a few warnings which we'll clean up but this should work on gazebo.
But we will fix https://github.com/fetchrobotics/fetch_ros/issues/102 asap.

We plan to release to Melodic next week, it will take a little until the next melodic sync and the debians are available.

closed 2019-01-27 by moriarty

varunagrawal, 2019-01-27:
@moriarty thanks for being on top of this! I may have been a bit harsh with my words before but I have high expectations of you guys and really appreciate the great work you are doing.

mikeferguson, 2019-01-27:
> @mikeferguson any updates on this?

Just FYI -- I'm not with Fetch since end of 2017, so I'm not the guy to ping on these in the future.

moriarty, 2019-04-05:
After having some issues with releasing for Melodic, I ended up doing a bunch of extra testing in Kinetic, and then decided that I should just release to Kinetic too.

But, we're only supporting Fetch in Gazebo for Kinetic. The drivers, firmware and actual robot haven't been tested in Kinetic.

moriarty, 2019-04-05:
http://repositories.ros.org/status_page/ros_kinetic_default.html?q=fetch_gazebo