fetchrobotics/fetch_gazebo #17: Simulated IMU

ruffsl, 2016-11-10:
I was testing out this package with gazebo and noticed that there was no imu messages being publish like for the real platform: http://docs.fetchrobotics.com/robot_hardware.html#imu

It'd be nice in general to add the IMUs in the gazebo models so I could test out projects like [cartographer](https://github.com/googlecartographer/cartographer_ros) that use senors fusion with IMUs in simulation. I suppose I could do a PR, but I'm not sure of the IMUs' origin wrt the robot's baselink and arm link.

ruffsl, 2017-02-01:
Related issue pending on this enhancement:
https://github.com/googlecartographer/cartographer_fetch/issues/2#event-941048389

moriarty, 2019-04-05:
Sorry @ruffsl this issue has been neglected a long time.

It's still very low priority but I've tagged @erelson and @ajzeller because they should be able to get the IMUs' origin wrt the robot's base link.

erelson, 2019-04-08:
For future reference, internal ticket: HP-2127

If anyone wants to make a PR with dummy values, demonstrating what components of the URDF the they think the IMUs should be linked to, that may make this happen sooner :)