fetchrobotics/fetch_gazebo #18: Add multiple Fetch Robots?

songrotek, 2016-12-26:
I am trying to add multiple Fetch robots in the gazebo environment, but it seems too difficult to make it work. Could you provide this feature? Thank you!

mikeferguson, 2016-12-27:
Unfortunately, this isn't something that is natively supported by Gazebo/ROS. I have seen some experimental stuff along these lines, but getting such a system to work reliably is likely a large development effort, which is currently not on our roadmap at Fetch Robotics. We do welcome contributions from the community, so I will leave this ticket open as a possible enhancement.

moriarty, 2019-01-27:
Just as an update, we know this is desired, but it's still not on our roadmap (specifically multiple Fetch Mobile Manipulators in Gazebo). In the meantime, I'll close this issue.

We know better multi-robot simulation is desired. We've needed to build up several private systems to simulate multiple freight robots (no arms) but at the moment these are very purpose built simulators for our commercial applications, not a general simulator like Gazebo or Stage.

closed 2019-01-27 by moriarty