fetchrobotics/fetch_gazebo #20: Mobile Manipulation Demo Experiences Several Trajectory Failures

DanielTakeshi, 2018-05-02:
I am trying to follow the tutorials on running the mobile robot demo.

http://docs.fetchrobotics.com/gazebo.html#running-the-mobile-manipulation-demo

However, the code doesn't seem to run the demo successfully where it is stated that:

> demo.py - this our specific demo which navigates the robot from the starting pose in Gazebo to the table, raises the torso, lowers the head to look at the table, and then runs perception to generate a goal for MoveIt. The arm will then grasp the cube on the table, tuck the arm and lower the torso. Once the robot is back in this tucked configuration, the navigation stack will be once again called to navigate into the room with the countertop where the robot will place the cube on the other table.

First, a few preliminaries: I'm using Ubuntu 14.04 and Ros Indigo, and have installed the simulator:

```
>$ sudo apt-get update
>$ sudo apt-get install ros-indigo-fetch-gazebo-demo
```
Running the playground seems to work:

```
daniel@daniel-ubuntu-mac:/opt/ros/indigo/share/fetch_gazebo$ roslaunch fetch_gazebo playground.launch
... logging to /home/daniel/.ros/log/ca0ab696-4e23-11e8-bb5b-7831c1b89008/roslaunch-daniel-ubuntu-mac-4627.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.121:46069/

SUMMARY
========

PARAMETERS
 * /arm_controller/follow_joint_trajectory/joints: ['shoulder_pan_jo...

// skipping a lot of parameters here

 * /head_controller/point_head/type: robot_controllers...
 * /publish_base_scan_raw/lazy: True
 * /robot/serial: ABCDEFGHIJKLMNOPQ...
 * /robot/version: 0.0.1
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 100.0
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /torso_controller/follow_joint_trajectory/joints: ['torso_lift_joint']
 * /torso_controller/follow_joint_trajectory/type: robot_controllers...
 * /use_sim_time: True

NODES
  /head_camera/
    crop_decimate (nodelet/nodelet)
    head_camera_nodelet_manager (nodelet/nodelet)
  /
    cmd_vel_mux (topic_tools/mux)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    prepare_robot (fetch_gazebo/prepare_simulated_robot.py)
    publish_base_scan_raw (topic_tools/relay)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    urdf_spawner (gazebo_ros/spawn_model)
  /head_camera/depth_downsample/
    points_downsample (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [4642]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ca0ab696-4e23-11e8-bb5b-7831c1b89008
process[rosout-1]: started with pid [4655]
started core service [/rosout]
process[gazebo-2]: started with pid [4679]
process[gazebo_gui-3]: started with pid [4683]
process[robot_state_publisher-4]: started with pid [4687]
[... 242 lines cut ...]
[INFO] [WallTime: 1525278041.088805] [293.587000] Picking object...
```

From the simulator, the first interesting thing is that the robot doesn't rotate correctly to start:

[image]

It was a bit stuck in this and rotating for a while, but as the debug messages above show, there is recovery behavior in this code. My machine is a bit old, so perhaps the slower the machine the more likely we run into these warning messages: `Map update loop missed its desired rate of 5.0000Hz... the loop actually took 0.2070 seconds`?

The real problem, however, is that once the Fetch rotates and raises its torso, the perception fails. From the simulator, it looks like the robot can't even see the cube.

[image]

I had to kill the program.

I ran this again but got a different issue; the robot was able to go to the second table --- but it didn't grasp the block, the image below was after the end (the block is out of view, sorry, but is still on the first table).

[image]

I guess there's two questions out of this:

- Is this kind of behavior expected for the simulator, even for a relatively fixed environment like this one?
- How do you  debug the `demo.py` code? As in, how do you figure out the locations where to command the Fetch to, increase robustness, etc?

Thanks for any help you may provide! Happy to also provide more info as needed.

mikeferguson, 2018-05-04:
Based on where the head is pointing in the second to last image, it would appear the fetch is not localized when it gets to the table. I did successfully run this code recently when I was validating/making the latest navigation release, so I don't think there is anything "broken" within the code.

If things aren't running fast enough, I could definitely see that being a problem -- there are a ton of loosely synchronized things going on here (physics simulation, controls, planning, sensing) and timing is essential.

It should also be noted that overall physics of the simulation are somewhat sketchy/suspect (This is true of pretty much all Gazebo sim). The demo works most of the time (85+%), but does occasionally fail -- and usually that is due to the physics rather than the actual navigation/manipulation code (which run much better on the real robot than the simulated one).

DanielTakeshi, 2018-05-08:
Thanks @mikeferguson  I think this makes sense

closed 2018-06-15 by DanielTakeshi