fetchrobotics/fetch_gazebo #22: Using gazebo7 with the Fetch's gazebo?

DanielTakeshi, 2018-06-11:
Hi @mikeferguson  et al,

Do you know if there is a way to install Fetch's gazebo stuff with gazebo 7? I have the usual Fetch setup with ROS Indigo and Ubuntu 14.04, but I was asking because I'm trying to use some code that's shared among different robots, and for the test worlds that we're using, we need gazebo 7 (since the "SDF" files have to be version 1.6, not 1.4 that I'm getting with gazebo2 now).

I was just inquiring because if I do a  `sudo apt-get install ros-indigo-fetch-gazebo-demo` or `sudo apt-get install ros-indigo-fetch-gazebo`, these commands require gazebo2 as a dependency, BUT I noticed this repository has a gazebo7 branch. Is it possible to use that version of the fetch simulator, somehow?

mikeferguson, 2018-06-12:
The gazebo2 branch is the "right" branch for Indigo out of the box (and debian installed).

You might be able to install a different gazebo and use the alternate branch for it -- but it's really not tested for such things. I know for a while I had gazebo5 (which shipped with Jade) on 14.04 with mostly Indigo sources and the gazebo5 branch of this repo.

The folks from Fetch might have other input -- I'm no longer at Fetch Robotics.

DanielTakeshi, 2018-06-15:
Thanks @mikeferguson

closed 2018-06-15 by DanielTakeshi

moriarty, 2019-03-22:
Thanks @mikeferguson.

@DanielTakeshi if you're still looking to use the Fetch with Gazebo7, the gazebo7 branch should be working, and recently updated, and I've done some testing on it.

It hasn't been released to Kinetic as a Debian package, because we're focusing on Indigo and Melodic for the real hardware, and didn't want the confusion of supporting the I, K, M in simulation but only I and M on hardware... However, I'm considering releasing the simulation to kinetic.

If you have the real hardware, then I recommend Indigo or Melodic. Currently most of the testing and effort targets Melodic. Indigo is just stable. But although the hardware works, we're still debugging issues with the simulation in Gazebo9.

And sorry, I never saw this issue until now because it was closed.
I was just notified about #48 and updated the README, and found this issue when searching closed issues.

DanielTakeshi, 2019-03-23:
@moriarty OK, thanks for the update.