fetchrobotics/fetch_gazebo #27: controllers are not started in fetch_gazebo

pholthau, 2018-11-14:
Hi,

When launching `roslaunch fetch_gazebo playground.launch` a number of services are not
started and therefore the preparation script hangs as well. The result is a non-functioning simulated robot (base and arm).

[image]

I think this might be a hiccup because I run fetch_gazebo on kinetic. But also, the issue does not occur 100% of the time but maybe 90%.

This is the log:

<details><summary><strong>output of roslaunch fetch_gazebo playground.launch</strong></summary>
roslaunch fetch_gazebo playground.launch 
... logging to /home/pholthaus/.ros/log/e8326c58-e7e9-11e8-9a5a-9cb6d0e2076f/roslaunch-rh-laptop-28376.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://192.168.1.66:46687/

SUMMARY
========

PARAMETERS
 * /arm_controller/follow_joint_trajectory/joints: ['shoulder_pan_jo...
 * /arm_controller/follow_joint_trajectory/type: robot_controllers...
 * /arm_controller/gravity_compensation/autostart: True
 * /arm_controller/gravity_compensation/root: torso_lift_link
 * /arm_controller/gravity_compensation/tip: gripper_link
 * /arm_controller/gravity_compensation/type: robot_controllers...
 * /arm_with_torso_controller/follow_joint_trajectory/joints: ['torso_lift_join...
 * /arm_with_torso_controller/follow_joint_trajectory/type: robot_controllers...
 * /base_controller/autostart: True
 * /base_controller/type: robot_controllers...
 * /bellows_controller/autostart: True
 * /bellows_controller/controlled_joint: bellows_joint
 * /bellows_controller/mimic_joint: torso_lift_joint
 * /bellows_controller/mimic_scale: 0.5
 * /bellows_controller/type: robot_controllers...
 * /gazebo/bellows_joint/position/d: 0.0
 * /gazebo/bellows_joint/position/i: 0.0
 * /gazebo/bellows_joint/position/i_clamp: 0.0
 * /gazebo/bellows_joint/position/p: 10.0
 * /gazebo/bellows_joint/velocity/d: 0.0
 * /gazebo/bellows_joint/velocity/i: 0.0
 * /gazebo/bellows_joint/velocity/i_clamp: 0.0
 * /gazebo/bellows_joint/velocity/p: 25.0
 * /gazebo/default_controllers: ['arm_controller/...
 * /gazebo/elbow_flex_joint/position/d: 0.1
 * /gazebo/elbow_flex_joint/position/i: 0.0
 * /gazebo/elbow_flex_joint/position/i_clamp: 0.0
 * /gazebo/elbow_flex_joint/position/p: 100.0
 * /gazebo/elbow_flex_joint/velocity/d: 0.0
 * /gazebo/elbow_flex_joint/velocity/i: 0.0
 * /gazebo/elbow_flex_joint/velocity/i_clamp: 0.0
 * /gazebo/elbow_flex_joint/velocity/p: 150.0
 * /gazebo/forearm_roll_joint/position/d: 0.1
 * /gazebo/forearm_roll_joint/position/i: 0.0
 * /gazebo/forearm_roll_joint/position/i_clamp: 0.0
 * /gazebo/forearm_roll_joint/position/p: 100.0
 * /gazebo/forearm_roll_joint/velocity/d: 0.0
 * /gazebo/forearm_roll_joint/velocity/i: 0.0
 * /gazebo/forearm_roll_joint/velocity/i_clamp: 0.0
 * /gazebo/forearm_roll_joint/velocity/p: 150.0
 * /gazebo/head_pan_joint/position/d: 0.0
 * /gazebo/head_pan_joint/position/i: 0.0
[... 501 lines cut ...]
[ INFO] [1542185616.629419763, 240.527000000]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1542185616.629518238, 240.527000000]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1542185616.629553601, 240.527000000]: MoveGroup context initialization complete

You can start planning now!

[ WARN] [1542185623.940113060, 244.175000000]: Timed out waiting for transform from base_link to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.101 timeout was 0.1.
</details>

moriarty, 2019-01-27:
This is because of the order of execution of launch files isn't guaranteed in ROS1 :disappointed: but there are ways around this, sorry about that.

We'll figure something out.

moriarty, 2019-03-19:
@pholthau is this still occuring? I've been launching, and launching, and launching while trying to tune the model for gazebo9, and haven't seen this issue

moriarty, 2019-04-05:
@pholthau I'm going to close this issue because I haven't been seeing it happen.

Feel free to open it up if it is still happening, and provide a little bit more information so that I can dig into why it's happening.

OS, ROS version, Package version, CPU & Hardware etc.

closed 2019-04-05 by moriarty