fetchrobotics/fetch_gazebo #30: [demo.py] the robot keeps rotating after moved near table

umhan35, 2019-02-14:
Hi, I am using Ubuntu 18.04 and ROS Melodic (to align with the version that the FetchIt robot has).

I followed the gazabo tutorial (https://docs.fetchrobotics.com/gazebo.html) and run the following:

    roslaunch fetch_gazebo playground.launch
    roslaunch fetch_gazebo_demo demo.launch

I expect the robot will move to the table and pick the blue cube, but after the robot moved near table keeps rotating it self. During 2 times, I waited 10 minutes before "Aborting because a valid control could not be found. Even after executing all recovery behaviors"

Rotating:
[image]

[image]

[image]

Here is the excerpt of the log:

```
[INFO] [1550152825.917689, 39.862000]: Waiting for basic_grasping_perception/find_objects...
[INFO] [1550152826.167008, 39.894000]: Moving to table...
[ WARN] [1550152935.337158025, 67.535000000]: Clearing costmap to unstuck robot (3.000000m).
[ WARN] [1550152980.132707802, 77.615000000]: Rotate recovery behavior started.
[ WARN] [1550153058.423329389, 95.251000000]: Invalid Trajectory 0.093694, 0.000000, 0.025957, cost: -1.000000
[ WARN] [1550153093.262487593, 103.608000000]: Clearing costmap to unstuck robot (3.000000m).
[ WARN] [1550153134.243491509, 113.969000000]: Clearing costmap to unstuck robot (3.000000m).
[ WARN] [1550153178.759782706, 125.168000000]: Clearing costmap to unstuck robot (3.000000m).
[ WARN] [1550153254.751576573, 143.775000000]: Clearing costmap to unstuck robot (3.000000m).
[ WARN] [1550153327.259470285, 162.218000000]: Rotate recovery behavior started.
[ WARN] [1550153366.278706368, 172.399000000]: Clearing costmap to unstuck robot (0.500000m).
[ERROR] [1550153366.753323531, 172.518000000]: Aborting because a valid control could not be found. Even after executing all recovery behaviors
[INFO] [1550153366.785093, 172.526000]: Raising torso...
Get block to pick...
[INFO] [1550153391.572698, 178.801000]: Picking object...
[WARN] [1550153392.673143, 178.988000]: ObjectManager: object0 not added yet
[WARN] [1550153392.681189, 178.992000]: ObjectManager: object1 not added yet
[WARN] [1550153392.688479, 178.993000]: ObjectManager: object2 not added yet
[WARN] [1550153392.695956, 178.993000]: ObjectManager: surface0 not added yet
[WARN] [1550153392.702579, 178.996000]: ObjectManager: surface1 not added yet
[INFO] [1550153393.662212, 179.207000]: Beginning to pick.
[ INFO] [1550153393.710886642, 179.216000000]: Planning attempt 1 of at most 1

```

Here is the full log:

```
$ roslaunch fetch_gazebo_demo demo.launch
... logging to /home/zhao/.ros/log/968350b0-3060-11e9-8d6a-04d3b0082825/roslaunch-zhao1804-15024.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zhao1804:45795/

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: base_link
 * /amcl/global_frame_id: map
 * /amcl/gui_publish_rate: 10.0
 * /amcl/kld_err: 0.01
 * /amcl/kld_z: 0.99
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 30
 * /amcl/laser_max_range: -1.0
 * /amcl/laser_min_range: -1.0
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
[... 882 lines cut ...]
[ INFO] [1550153400.624258590, 180.820000000]: Manipulation plan 2 failed at stage 'reachable & valid pose filter' on thread 2
[ INFO] [1550153400.626616510, 180.820000000]: Manipulation plan 6 failed at stage 'reachable & valid pose filter' on thread 0
[ INFO] [1550153400.630235842, 180.820000000]: Manipulation plan 15 failed at stage 'reachable & valid pose filter' on thread 1
[ INFO] [1550153400.639196048, 180.821000000]: Manipulation plan 10 failed at stage 'reachable & valid pose filter' on thread 3
[ INFO] [1550153400.649360544, 180.823000000]: Manipulation plan 14 failed at stage 'reachable & valid pose filter' on thread 2
[ INFO] [1550153400.656643976, 180.823000000]: Manipulation plan 3 failed at stage 'reachable & valid pose filter' on thread 0
[ INFO] [1550153400.657404738, 180.823000000]: Manipulation plan 8 failed at stage 'reachable & valid pose filter' on thread 1
[ INFO] [1550153400.668111743, 180.826000000]: Manipulation plan 16 failed at stage 'reachable & valid pose filter' on thread 3
[ INFO] [1550153400.679405593, 180.826000000]: Manipulation plan 12 failed at stage 'reachable & valid pose filter' on thread 2
[ INFO] [1550153400.687058933, 180.827000000]: Manipulation plan 13 failed at stage 'reachable & valid pose filter' on thread 1
[ INFO] [1550153400.690214022, 180.827000000]: Manipulation plan 17 failed at stage 'reachable & valid pose filter' on thread 0
[ INFO] [1550153400.694703205, 180.827000000]: Manipulation plan 4 failed at stage 'reachable & valid pose filter' on thread 2
[ INFO] [1550153400.696004253, 180.827000000]: Manipulation plan 9 failed at stage 'reachable & valid pose filter' on thread 3
[ WARN] [1550153400.708756540, 180.830000000]: All supplied grasps failed. Retrying last grasp in verbose mode.
[ INFO] [1550153400.708940172, 180.830000000]: Re-added last failed plan for pipeline 'pick'. Queue is now of size 1
[ INFO] [1550153400.711116214, 180.830000000]: IK failed
[ INFO] [1550153400.719230453, 180.830000000]: IK failed
[ INFO] [1550153400.722598089, 180.830000000]: IK failed
[ INFO] [1550153400.722787301, 180.830000000]: Sampler failed to produce a state
[ INFO] [1550153400.722872173, 180.830000000]: Manipulation plan 9 failed at stage 'reachable & valid pose filter' on thread 1
[ INFO] [1550153400.727197412, 180.830000000]: Pickup planning completed after 0.137375 seconds
[ERROR] [1550153400.736623, 180.835000]: Pick failed in the planning stage, try again...
^C

```

moriarty, 2019-02-14:
Sounds like the same issue reported here https://github.com/fetchrobotics/fetch_ros/issues/102

@narora1 @safrimus

moriarty, 2019-04-05:
We've made some improvements but I think this is still a bit of an issue.

I've added Help Wanted and More Info Needed, similar to #36

xujingren, 2019-07-04:
@umhan35 Hi, it might work better if you reduce the inflation_radius specified in costmap_common.yaml under fetch_navigation/config. (the default value is 0.7m)
In your case, the robot has to be close to the table to pick up the cube, while large inflation_radius keeps the robot away from the obstacles, and the robot starts rotation recovery if it is within the inflation layer of the table.

umhan35, 2019-07-04:
@xujingren We actually have another map and move_base config that works in our test course but not in Gazebo 9. The root cause is #102. We tested on Ubuntu 14.04 with Gazebo 2 and it works.