fetchrobotics/fetch_gazebo #31: [pick_place_demo.py] cannot grasp the cube

umhan35, 2019-02-14:
Related to issue #30, I tried to run the pick_place_demo.py without the navigation part:

    roslaunch fetch_gazebo pickplace_playground.launch
    roslaunch fetch_gazebo_demo pick_place_demo.launch

It seems there are some error detecting the cube.

Grasping:

[image]

Picking:

[image]

Here is the full log:

$ roslaunch fetch_gazebo_demo pick_place_demo.launch 
... logging to /home/zhao/.ros/log/e097f2ea-3066-11e9-8d6a-04d3b0082825/roslaunch-zhao1804-28988.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zhao1804:42179/

SUMMARY
========

PARAMETERS
 * /basic_grasping_perception/gripper/approach/desired: 0.15
 * /basic_grasping_perception/gripper/approach/min: 0.145
 * /basic_grasping_perception/gripper/finger_depth: 0.02
 * /basic_grasping_perception/gripper/gripper_tolerance: 0.05
 * /basic_grasping_perception/gripper/retreat/desired: 0.15
 * /basic_grasping_perception/gripper/retreat/min: 0.145
 * /basic_grasping_perception/gripper/tool_to_planning_frame: 0.18
 * /move_group/allow_trajectory_execution: True
 * /move_group/allowed_execution_duration_scaling: 1.2
 * /move_group/allowed_goal_duration_margin: 0.5
 * /move_group/arm/longest_valid_segment_fraction: 0.05
 * /move_group/arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/arm/projection_evaluator: joints(shoulder_p...
 * /move_group/arm_with_torso/longest_valid_segment_fraction: 0.05
 * /move_group/arm_with_torso/planner_configs: ['SBLkConfigDefau...
 * /move_group/arm_with_torso/projection_evaluator: joints(torso_lift...
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'default': True...
 * /move_group/gripper/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /move_group/planner_configs/TRRTkConfigDefault/type: geometric::TRRT
 * /move_group/planning_plugin: ompl_interface/OM...
 * /move_group/planning_scene_monitor/publish_geometry_updates: True
 * /move_group/planning_scene_monitor/publish_planning_scene: True
 * /move_group/planning_scene_monitor/publish_state_updates: True
 * /move_group/planning_scene_monitor/publish_transforms_updates: True
 * /move_group/request_adapters: default_planner_r...
 * /move_group/start_state_max_bounds_error: 0.1
[... 226 lines cut ...]
[ INFO] [1550155464.148302253, 19.709000000]: Manipulation plan 1 failed at stage 'reachable & valid pose filter' on thread 3
[ INFO] [1550155464.149319629, 19.709000000]: Manipulation plan 15 failed at stage 'reachable & valid pose filter' on thread 0
[ INFO] [1550155464.155938859, 19.709000000]: Manipulation plan 3 failed at stage 'reachable & valid pose filter' on thread 0
[ INFO] [1550155464.163443144, 19.709000000]: Manipulation plan 14 failed at stage 'reachable & valid pose filter' on thread 3
[ INFO] [1550155464.180421518, 19.710000000]: Manipulation plan 16 failed at stage 'reachable & valid pose filter' on thread 3
[ INFO] [1550155464.183416335, 19.710000000]: Manipulation plan 8 failed at stage 'reachable & valid pose filter' on thread 0
[ INFO] [1550155464.189920666, 19.711000000]: Manipulation plan 17 failed at stage 'reachable & valid pose filter' on thread 0
[ INFO] [1550155465.402118125, 19.831000000]: Manipulation plan 11 failed at stage 'reachable & valid pose filter' on thread 1
[ INFO] [1550155465.412926059, 19.832000000]: Manipulation plan 9 failed at stage 'reachable & valid pose filter' on thread 1
[ INFO] [1550155465.422259278, 19.834000000]: Manipulation plan 4 failed at stage 'reachable & valid pose filter' on thread 1
[ INFO] [1550155471.756547482, 20.682000000]: Found a contact between 'object0' (type 'Object') and 'r_gripper_finger_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1550155471.756716465, 20.682000000]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1550155471.756875507, 20.682000000]: Start state appears to be in collision with respect to group arm
[ INFO] [1550155471.757233064, 20.682000000]: Found a valid state near the start state at distance 0.086531 after 0 attempts
[ INFO] [1550155471.758017345, 20.682000000]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1550155471.758566091, 20.682000000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1550155471.773592136, 20.684000000]: RRTConnect: Created 5 states (3 start + 2 goal)
[ INFO] [1550155471.773683964, 20.684000000]: Solution found in 0.015258 seconds
[ INFO] [1550155471.824327334, 20.691000000]: SimpleSetup: Path simplification took 0.050551 seconds and changed from 4 to 2 states
[ INFO] [1550155471.830048520, 20.692000000]: Planning adapters have added states at index positions: [ 0 ]
[ INFO] [1550155471.837971940, 20.692000000]: Found successful manipulation plan!
[ INFO] [1550155473.663829588, 20.963000000]: Manipulation plan 13 failed at stage 'approach & translate' on thread 0
[ INFO] [1550155474.116754124, 21.039000000]: Manipulation plan 12 failed at stage 'approach & translate' on thread 3
[ INFO] [1550155474.116946183, 21.039000000]: Pickup planning completed after 10.014108 seconds
^C

moriarty, 2019-02-14:
Thanks, how reproducible is this?

Does it fail every time? Or sometimes pass as mentioned in this issue #20

umhan35, 2019-02-15:
I tried like 5 times, it was never successful.  One thing is that I am using integrated GPU (Intel® HD Graphics 620).

Here is my environment setup:

# Environment Setup

Make sure Ubuntu 18.04 and ROS Melodic are installed. We are using this version because it is also installed on the robot and it works.

(Currently only ROS Indigo is supported as Debian packages, but there are git branches for Ubuntu 18.04/ROS Melodic and both are installed on the onboard computer of the robot.)

## Installing [fetch_ros](https://github.com/fetchrobotics/fetch_ros) & [fetch_gazebo](https://github.com/fetchrobotics/fetch_gazebo) from source

Install the dependencies for Fetch Robotics:

    sudo apt-get update
    sudo apt install ros-melodic-robot-controllers ros-melodic-costmap-2d ros-melodic-moveit ros-melodic-rgbd-launch

Create a catkin workspace: http://wiki.ros.org/catkin/Tutorials/create_a_workspace

In the carkin workspace `src` directory, run:

    git clone https://github.com/fetchrobotics/fetch_ros.git
    git clone https://github.com/fetchrobotics/fetch_gazebo.git

In your carkin workspace `root` directory, run

    catkin_make

## Test your installation in an empty world

    roslaunch fetch_gazebo simulation.launch

## Gazebo demo

Install dependencies:

    sudo apt install ros-melodic-map-server ros-melodic-simple-grasping ros-melodic-amcl ros-melodic-move-base

demo:

    roslaunch fetch_gazebo pickplace_playground.launch
    roslaunch fetch_gazebo_demo pick_place_demo.launch

umhan35, 2019-02-15:
I also made changes mentioned in PR #26.

moriarty, 2019-02-28:
We have a new Intern: @ejichen who is working on updating our demos with the Fetch in Simulation and on the Real Robot.

She started this week and will work on a new cube picking demo.

I'll assign this to her to look into.

1. I've noticed in Gazebo9, the lighting is much more accurate than it was in the past. We may need to update our simulation world to increase the lighting to improve the ability to detect the cube.
2. There are a number of other tickets related to the control of the base/gripper/arm etc. I think we can consulate them all into one ticket to focus our efforts.

ejichen, 2019-03-13:
Hi @umhan35,
I tried to reproduce the issue and the situation showed in the [video](https://youtu.be/q56RsyfWfPw) might be similar to your problem.

On the first try, the Gazebo window shows that the robot flickered a little bit after it moved the arm. The movement shifted the robot, resulting in the mismatch of the robot frame and the planning scene frame (shows in the rviz window). 

The second try didn't succeed because the previous movement made some changes to the position of the cube. However, if you let the simulation run for a while, you could see the pick and place demo succeed with the third try (since the script is a while loop of pick and place).

moriarty, 2019-03-20:
@umhan35 I've made two URDF changes.

1. Quick Fix: https://github.com/fetchrobotics/fetch_gazebo/issues/37#issuecomment-473019867
2. Better Fix: https://github.com/fetchrobotics/fetch_gazebo/issues/37#issuecomment-473807444

@ejichen has been using **1.** which is a quick fix, but I will make a PR with **2.** which adds the caster wheels to the model

moriarty, 2019-03-21:
Actually, I before committing the changes, I went to record before and after videos.

And surprisingly the fetch was able to grasp the cube, using the ccb5ad3f40cedc2944a9b3d6fab8d75239a9997b and the fetch-ros from debians https://github.com/fetchrobotics/fetch_ros/releases/tag/0.8.0.

https://youtu.be/izZZFZgtJYo

I didn't use **fetch_ros** from source- the versions from source upgraded the fetch_ikfast_plugin https://github.com/fetchrobotics/fetch_ros/pull/107 to fix deprecation warnings with the next version of moveit.

umhan35, 2019-03-21:
I'd say the behavior is not consistent. I am not sure if it is a Gazebo issue.

moriarty, 2019-03-21:
The behaviour isn’t expected to be consistent, it’s running MoveIt! with random sampling based motion planners.

But, if we make the URDF changes to stablize the base (the real Robot is very stable and doesn‘t wiggle or tilt when moving the arm) then it should behave closer to the real thing. In #37, I’ve posted videos of Indigo & Kinetic and the inertia/CoM values visually look different- despite not being changed.

PR #49 (2019-03-25): fetch_gazebo: add casters to fetch.gazebo.xacro
This adds 4 casters to the gazebo model, and updates the physics params.

This should fix #31 or at least make it better. It should also fix #35. This is related to #37.

I launched this twice and it grasped the cube both times, so it's much better than it was.
But the values still need further tuning for navigation, and e.g. #30
commit 20bf7c81 (2019-03-25): fetch_gazebo: add casters to fetch.gazebo.xacro (#49)

This adds 4 casters to the gazebo model, and updates the physics params.

This should fix #31 and fix #35 

This is related to #37
closed 2019-03-25 by moriarty