fetchrobotics/fetch_gazebo #33: Feature: Add option to start low-quality simulation for fetchit_challenge

ben-fido, 2019-02-27:
Hello, great job with the new simulation. It looks very nice.

**Issue description:**
I start the simulation environment like this:
`roslaunch fetchit_challenge main.launch`
I wonder if all the extra objects make it slower to render or simulate on my computer.

**Feature request:**
Add a parameter to start a low-quality simulation.
This could disable spawning of objects like the carpet, pallet racks, barrels and factory walls.

ejichen, 2019-02-27:
seems reasonable. - from @moriarty

moriarty, 2019-02-28:
@ben-fido that above comment was from me, I was just sitting in front of another computer.

We're working with TheConstruct to make the simulation environment ready and they used their existing environment with minimal changes. The environment can be simplified, ans we're setting up a small space in our offices which doesn't have the extra elements.

Next week sometime @narora1, @cjds, @ajzeller and @velveteenrobot will meet with some representatives from SICK and EandM to finalize the location of the SICK Lector, the layout and do a walk through... I think someone can get the room dimensions of the room we've setup, so that the small 10x10 environment is within just a smaller empty room.

@RDaneelOlivav- sound reasonable?

RDaneelOlivav, 2019-02-28:
@moriarty, @ben-fido  , we have recently been working on a simpler version of the simulation, with as few elements as possible ( @mmwise suggested it after a test detecting that it was laggy ).
I've also corrected the tables not appearing issue and also one of the gears. It was a materials transparency value that got there accidentally. SO now through the cameras, it should appear every element.
We also changed a bit the sponsor panel.
Let me know your opinion about these changes and suggestions for more improvements.
We will try also to tweak the physics to get the fastest experience. I'll post a pull request this week for these changes.

[image]

RDaneelOlivav, 2019-02-28:
Also, yeah, as soon as we define more and more the challenge and how everything will be done, we will be able to simplify more and more.

Another idea could be also to make a low poly, simplified version even further to maximize the tests.
Tell me what you think about that ;).

moriarty, 2019-04-05:
:+1: thanks I'm closing this one.

The low poly simplified version would be nice but there is already a lot todo so we can just keep that in mind for the future.

closed 2019-04-05 by moriarty