fetchrobotics/fetch_gazebo #34: Bug: Simulation fetchit_challenge looks different in Gazebo & RDS

ben-fido, 2019-02-27:
**Cloud simulator:** RDS (ROS Development Studio), Ubuntu 18.04, ROS Melodic
**Local computer:** Ubuntu 18.04, ROS Melodic, Gazebo 9, NVidia GPU

Has anyone compared the FetchIt Challenge simulator running on RDS with installing on local computer?

**Bug description:**
I find there are various differences, like objects that are not spawned or maybe they are just hidden.
 - The two bolts are blue on RDS but orange on local computer.
 - The two gears. On RDS there is a red and a green gear. My local computer is missing the large green gear.
 - The 3 tables are transparent on local computer.
 - The Schunk machine has different appearance on local computer, like half of it is white instead of grey.
 - The lights are rendered differently. RDS looks more like Gazebo2. Running Gazebo9 on local computer, objects are either too bright or too dark.
 - Non-essential objects like some barrels and a box are missing on local computer.

Here are some screenshots:

[image]

[image]

[image]

[image]

[image]

moriarty, 2019-02-27:
The local version is probably slightly more up to date, because the screws in the bin, shouldn't be the same colour as the bin.

Although the local version lighting looks too bright.

@RDaneelOlivav

RDaneelOlivav, 2019-02-28:
@moriarty , Just seen this, in my next pull request the floating tables should be resolved ;). How can I see the issues assigned to me ?

RDaneelOlivav, 2019-02-28:
@ben-fido Also I've created a version with lower lights to accomodate to different needs. The Gazebo simulator renders slightly different the lights in the client, so you will see it different from the Cloud Version. 
But the rendering of the cameras and all the sensors is EXACTLY the same. So please check the cameras also for the one that suits best your needs. I'll post all these changes ASAP.

moriarty, 2019-03-14:
@ben-fido yes, we're working with @RDaneelOlivav to iterate over the environment and sync everything... We're also building up the physical arena on site which now that it's almost build @velveteenrobot will share all the new measurements and we'll need another iteration of the environment.

Yesterday we did some testing with the real robot, and the real parts, and real Schunk machine and @ajzeller will give one of the gears a longer shaft to increase the tolerance when inserting it into the machine.

We may also update the handle of the machine, the robot is able to grasp the handle and open the door, but the robot needs to do a top down grasp on the current handle, and that essentially wastes some degrees of freedom. I'm looking at changing the handle to make the a horizontal grasp possible and linear motion easier. We'll discuss this with @mmwise and @rctoris.

ejichen, 2019-03-15:
Also, there is an invisible gear on the desk.
[image]

ejichen, 2019-03-15:
Nevermind. The latest version in git fixed the issue.
[image]

moriarty, 2019-03-15:
@velveteenrobot has measured our arena environment in house and after the next update I should re-bloom this package.

moriarty, 2019-04-05:
@ben-fido I'll close this out for now, we've split this up into multiple new tickets:

#47 Additional Kits
#61 Update SCHUNK Machine
#73 Update tables and walls
#74 Update colours of parts

In all cases, the simulation as it is, although different in ROSDS & Gazebo, is already near enough to test components of the challenge code and shouldn't be a blocking issue.

As pointed out by @RDaneelOlivav the how Gazebo renders the scene and lighting may _appear_ different in the view but how that translates into sensor data will not be different. In my experience Gazebo looks differently from machine to machine with the exact same version just based on the GPU in the machine, but the sensor data as seen by the Robot will look the same.

closed 2019-04-05 by moriarty