fetchrobotics/fetch_gazebo #40: Construct sim with fetch throws an exception

narora1, 2019-03-17:
I tried getting fetch working on 18.04, melodic and with gazebo 9 but got this exception.

[image]

Also is there a way I can make edits to the fetch robot model ?

RDaneelOlivav, 2019-03-18:
I'll check this today for sure, keep you posted.

RDaneelOlivav, 2019-03-18:
Ok, I'm sharing with you a ROSJECT that I have launched and works:
LINK--> Just click it and you will have it on your space: http://www.rosject.io/l/8361c2a/

[image]

You just have to launch the main_simple.launch from the package fetchit_challenge from the simulations panel ( select launch file ) and you should get the simulation running.

Any issue please report it so we can resolve it ASAP ;).

narora1, 2019-03-19:
[image]
It didn't work it timed out waiting for simulation.

RDaneelOlivav, 2019-03-19:
Ok let me have a look at that in depth because as far as I've tested it worked, save included.

RDaneelOlivav, 2019-03-19:
Ok we have been debugging and as far as we see to debug the issue we should have access to your system when running. Can you contact me when you make a test so that we can debug this issue? Because we can't reproduce the issue here for some reason. 
@narora1 
duckfrost@gmail.com is my gmail account. Send me an email or a hangouts message and we will be there. We will solve the issue but having your system running and seeing what happens will help a lot for sure.

RDaneelOlivav, 2019-03-19:
What OS and browser are you using? @narora1

narora1, 2019-03-20:
Initially it was Firefox and Ubuntu 14.04. The one with the timeout was MacOs on Safari

narora1, 2019-03-20:
[image]

It has been 20minutes and it is stuck loading. This is similar to what happened before until it timed out. This time I was using Firefox and ubuntu 14.04. Is there a preferred browser that I should test this on?

RDaneelOlivav, 2019-03-22:
Hi!
We fixed some issues that we detected due to this, could you retry again, please.
You shouldn't have this error anymore.

Please repost if you have any issue and if not post an image of the ROJect running to see that everything is running ;).

narora1, 2019-03-28:
[image]

works now!!

closed 2019-03-28 by narora1

RDaneelOlivav, 2019-03-28:
Great to hear it! Remember to pull the fetch_gazebo git once in a while if you want to stay up to date with the improvements and the latest version of the simulation ;)