fetchrobotics/fetch_gazebo #47: Additional Kits in the Gazebo World [from email]

moriarty, 2019-03-21:
> How many kit caddies will be place on the kit station table? The simulation only has one but the rulebook says “Kits will be approximately evenly spaced and not touching”, which means there will be several of them. If that’s the case, could you modify the Gazebo world to reflect that?

@umhan35 asked via email.

I just checked the arena, and 3 fit nicely on a table to start.

After 2 have been processed (either finished and delivered or dropped), the kit table will be replenished. But in the Gazebo Model, I think 3 will be fine, as two complete deliveries are required for winning a Fetch.

moriarty, 2019-03-21:
[image]

RDaneelOlivav, 2019-04-05:
So I suppose two more trays have to be created and placed in the simulation, a grey one and a yellow one. Is that it?

moriarty, 2019-04-05:
I just commented on #73 maybe add all the colours we have.

https://www.amazon.com/Classroom-Supplies-Assorted-Stackable-Plastic/dp/B074Q2Y7WR?ref_=fsclp_pl_dp_12

- Blue
- Green
- Yellow
- Red
- Orange
- Purple

And we ordered two more from a different supplier to check they were similar enough for object recognition (they are):
- Grey
- Green

moriarty, 2019-04-05:
In the default simulation, just spawn 3 for now like the photo above.

We assume teams know how to spawn an additional model if they want to test with other colours.

RDaneelOlivav, 2019-04-08:
Ok, will start ASAP ;).

RDaneelOlivav, 2019-04-11:
Made a Pull Request to implement this.
https://github.com/fetchrobotics/fetch_gazebo/pull/76

closed 2019-04-24 by moriarty