fetchrobotics/fetch_gazebo #48: Problem: gazebo7 with joint_handle.h and plugin.cpp

zhaolongkzz, 2019-03-22:
Hello. Thanks for your code of FETCH. 
And I run it in my computer with ubuntu16.04 and ROS kinetic, gazebo7.14.
Then I encounter some problem below, I think it belongs to the version of gazebo.

```bash
In file included from /home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/src/plugin.cpp:40:0:
/home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/include/fetch_gazebo/joint_handle.h: In member function ‘virtual double gazebo::JointHandle::getPosition()’:
/home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/include/fetch_gazebo/joint_handle.h:131:46: error: ‘class gazebo::physics::Joint’ has no member named ‘Position’
       return angles::normalize_angle(joint_->Position(0));
                                              ^
/home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/include/fetch_gazebo/joint_handle.h:133:20: error: ‘class gazebo::physics::Joint’ has no member named ‘Position’
     return joint_->Position(0);
                    ^
/home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/include/fetch_gazebo/joint_handle.h: In member function ‘virtual double gazebo::JointHandle::getPositionMin()’:
/home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/include/fetch_gazebo/joint_handle.h:157:20: error: ‘class gazebo::physics::Joint’ has no member named ‘LowerLimit’
     return joint_->LowerLimit(0);
                    ^
/home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/include/fetch_gazebo/joint_handle.h: In member function ‘virtual double gazebo::JointHandle::getPositionMax()’:
/home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/include/fetch_gazebo/joint_handle.h:163:20: error: ‘class gazebo::physics::Joint’ has no member named ‘UpperLimit’
     return joint_->UpperLimit(0);
                    ^
/home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/src/plugin.cpp: In member function ‘virtual void gazebo::FetchGazeboPlugin::Init()’:
/home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/src/plugin.cpp:97:40: error: ‘class gazebo::physics::World’ has no member named ‘SimTime’
   prevUpdateTime = model_->GetWorld()->SimTime();
                                        ^
/home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/src/plugin.cpp:110:71: error: conversion from ‘urdf::JointConstSharedPtr {aka boost::shared_ptr<const urdf::Joint>}’ to non-scalar type ‘std::shared_ptr<const urdf::Joint>’ requested
     std::shared_ptr<const urdf::Joint> urdf_joint = urdfmodel.getJoint((*it)->GetName());
                                                                       ^
/home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/src/plugin.cpp: In member function ‘void gazebo::FetchGazeboPlugin::OnUpdate(const gazebo::common::UpdateInfo&)’:
/home/zzl/fetch_ws/src/fetch_gazebo/fetch_gazebo/src/plugin.cpp:138:47: error: ‘class gazebo::physics::World’ has no member named ‘SimTime’
   common::Time currTime = model_->GetWorld()->SimTime();
                                               ^
fetch_gazebo/fetch_gazebo/CMakeFiles/fetch_gazebo_plugin.dir/build.make:62: recipe for target 'fetch_gazebo/fetch_gazebo/CMakeFiles/fetch_gazebo_plugin.dir/src/plugin.cpp.o' failed
make[2]: *** [fetch_gazebo/fetch_gazebo/CMakeFiles/fetch_gazebo_plugin.dir/src/plugin.cpp.o] Error 1
CMakeFiles/Makefile2:2965: recipe for target 'fetch_gazebo/fetch_gazebo/CMakeFiles/fetch_gazebo_plugin.dir/all' failed
make[1]: *** [fetch_gazebo/fetch_gazebo/CMakeFiles/fetch_gazebo_plugin.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

closed 2019-03-22 by zhaolongkzz

moriarty, 2019-03-22:
you've closed this issue? were you on the wrong branch?

moriarty, 2019-03-22:
I assume this didn't build because of the default branch in GitHub being for Gazebo9. I've updated the README.

zhaolongkzz, 2019-03-23:
Thanks, @moriarty I found I have git the wrong branch. And gazebo7 works well.