fetchrobotics/fetch_gazebo #56: gearbox_top and gearbox_largegear fall through the world

moriarty, 2019-03-27:
Related to #50 and #38 

@RDaneelOlivav the simulation environment is looking good but there is still some fine tuning which needs to be done.

the large gear and gearbox top, fall through the world

RDaneelOlivav, 2019-03-27:
Ok I'll see into it right now ;)

RDaneelOlivav, 2019-03-27:
Ok I've seen where is the issue. I didn't test the STL files, and they have a super big scale and are decentered. Solving it now ;). Just rescaling in the sdf gave me the hint, see the white box that doesnt correspond even if its rescaled?:

[image]

I'll Fork & Branch ;)

velveteenrobot, 2019-03-27:
I may change the large gear file again if we add a chamfer to the end of the shaft. @RDaneelOlivav so you can hold off on that one.

RDaneelOlivav, 2019-03-27:
@velveteenrobot Don't worry, if you make changes we do another branch and changes, but this change has to be made because otherwise the simulation is broken ;). Just Tag me in that new change and I'll go into it ;).

RDaneelOlivav, 2019-03-27:
@velveteenrobot By the way, maybe, to allow you to make faster changes and not have to wait that I correct scales and centers, I can post a small video tutorial on how to make those corrections yourself in Blender ;).

commit 1d681fbc (2019-03-27): [FetchIt!] fix the stl collision scales (#60)

Added rescaled stls with corrected origins that generated issue that seems they were no collisions but in fact they were decentered and huge.

This fixes #56
commit 0481932e (2019-03-28): [FetchIt!] fix the stl collision scales (#60)

Added rescaled stls with corrected origins that generated issue that seems they were no collisions but in fact they were decentered and huge.

This fixes #56
commit 5510c3e3 (2019-03-28): [FetchIt!] fix the stl collision scales (#60)

Added rescaled stls with corrected origins that generated issue that seems they were no collisions but in fact they were decentered and huge.

This fixes #56
closed 2019-03-27 by moriarty