fetchrobotics/fetch_gazebo #61: Update SCHUNK Machine

moriarty, 2019-03-27:
@dekent has emailed asking about the SCHUNK Machine.

It was originally added to Gazebo after we only saw the CAD files, and didn't know how it actually looked but we knew it had a Logo on it.

Here is a photo of the actual machine, next to a shelf.
The Shelves will have other logos on them, from SICK, EandM, Fetch and The Construct Sim.

[image]
[image]

This was one of our old demo warehouse shelves. I've cut it down to match the height of the Schunk machine.

They can be disassembled and then they flat-pack for shipping, so we will use the same shelves in Montreal.

We don't have hard covers for the sides yet, but lets assume they'll look similar to the Schunk machine. They will be made out of some stiff material (likely foam-core board) and attach sides with "velcro".
The other competition co-sponsors will place their logos on these shelf sides.

I just took a photo of the back to show that it was a shelf. But the version for gazebo does not need to model the inside, it can simply be a solid rectangle.

_Originally posted by @moriarty in https://github.com/fetchrobotics/fetch_gazebo/pull/32#issuecomment-465380300_

RDaneelOlivav, 2019-04-05:
What are the atsk here exactly? Add the action server that @velveteenrobot did for the real shunk? https://github.com/fetchrobotics/fetch_gazebo/pull/72.
I posted there that we would need what eahc action does in the real shunk to simulated in the simulated SHUNK machine so that it work in the exact same way ( or as close as it has to be )

Add the panels exactly in the same format as they are here?

moriarty, 2019-04-05:
Good point, I was just thinking cosmetically making the machine look closer to the real thing... The machine in sim has the logo on the machine, but the machine is actually metal, with the logo under the machine.

For making the machine work in sim, that can be another issue where #72 from @velventeenrobot is hooked up to a simple controller.

moriarty, 2019-06-05:
This is fixed with #89 I believe.

closed 2019-06-05 by moriarty