fetchrobotics/fetch_gazebo #66: One error

Zephyr06, 2019-03-31:
In file included from /opt/ros/melodic/include/actionlib/server/handle_tracker_deleter.h:42:0,
                 from /opt/ros/melodic/include/actionlib/server/action_server.h:51,
                 from /opt/ros/melodic/include/actionlib/server/simple_action_server.h:42,
                 from /opt/ros/melodic/include/robot_controllers_interface/controller_manager.h:36,
                 from /home/zephyr/catkin_ws/src/fetch_gazebo/fetch_gazebo/src/plugin.cpp:39:
/opt/ros/melodic/include/actionlib/destruction_guard.h: In member function ‘void actionlib::DestructionGuard::destruct()’:
/opt/ros/melodic/include/actionlib/destruction_guard.h:62:80: error: no matching function for call to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>::subsecond_duration(float)’
    count_condition_.timed_wait(lock, boost::posix_time::milliseconds(1000.0f));
                                                                             ^
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
                 from /usr/local/include/boost/date_time/posix_time/ptime.hpp:12,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:12,
                 from /usr/local/include/boost/thread/thread_time.hpp:11,
                 from /usr/local/include/boost/thread/detail/platform_time.hpp:11,
                 from /usr/local/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/local/include/boost/thread/condition_variable.hpp:16,
                 from /usr/local/include/boost/thread/condition.hpp:13,
                 from /opt/ros/melodic/include/actionlib/server/simple_action_server.h:40,
                 from /opt/ros/melodic/include/robot_controllers_interface/controller_manager.h:36,
                 from /home/zephyr/catkin_ws/src/fetch_gazebo/fetch_gazebo/src/plugin.cpp:39:
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note: candidate: template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<Functor>, void>::type*)
     explicit subsecond_duration(T const& ss,
              ^~~~~~~~~~~~~~~~~~
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note:   template argument deduction/substitution failed:
/usr/local/include/boost/date_time/time_duration.hpp: In substitution of ‘template<class T> boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<T> >::type*) [with T = float]’:
/opt/ros/melodic/include/actionlib/destruction_guard.h:62:80:   required from here
/usr/local/include/boost/date_time/time_duration.hpp:285:14: error: no type named ‘type’ in ‘struct boost::enable_if<boost::is_integral<float>, void>’
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
                 from /usr/local/include/boost/date_time/posix_time/ptime.hpp:12,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:12,
                 from /usr/local/include/boost/thread/thread_time.hpp:11,
                 from /usr/local/include/boost/thread/detail/platform_time.hpp:11,
                 from /usr/local/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/local/include/boost/thread/condition_variable.hpp:16,
                 from /usr/local/include/boost/thread/condition.hpp:13,
                 from /opt/ros/melodic/include/actionlib/server/simple_action_server.h:40,
                 from /opt/ros/melodic/include/robot_controllers_interface/controller_manager.h:36,
                 from /home/zephyr/catkin_ws/src/fetch_gazebo/fetch_gazebo/src/plugin.cpp:39:
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate: boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>::subsecond_duration(const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>&)
   class BOOST_SYMBOL_VISIBLE subsecond_duration : public base_duration
                              ^~~~~~~~~~~~~~~~~~
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note:   no known conversion for argument 1 from ‘float’ to ‘const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>&’
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate: boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>::subsecond_duration(boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>&&)
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note:   no known conversion for argument 1 from ‘float’ to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>&&’
CMakeFiles/fetch_gazebo_plugin.dir/build.make:62: recipe for target 'CMakeFiles/fetch_gazebo_plugin.dir/src/plugin.cpp.o' failed
make[2]: *** [CMakeFiles/fetch_gazebo_plugin.dir/src/plugin.cpp.o] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/fetch_gazebo_plugin.dir/all' failed
make[1]: *** [CMakeFiles/fetch_gazebo_plugin.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

I can't understand this error at all, so any suggestions? Thanks a lot!

Zephyr06, 2019-03-31:
After git clone, and use catkin_make_isolated, it reports it.

moriarty, 2019-03-31:
It's happening when it tries to include the `robot_controllers_interface/controller_manager.h`

And it's not happening on the build farm: http://build.ros.org/job/Mbin_uB64__fetch_simulation__ubuntu_bionic_amd64__binary/

Are the rest of your system packages up to date?

moriarty, 2019-04-04:
I'm going to close this for now. Let me know if it is still an issue.

closed 2019-04-04 by moriarty