fetchrobotics/fetch_gazebo #69: When catkin_make fetch_gazebo non-homogeneous workspace error appears

RDaneelOlivav, 2019-04-04:
When compiling the entire fetch_gazebo git it gives the following error:

`-- ~~  - fetch_simulation (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'fetch_simulation' contains non standard content. Use the content of the following file instead: /home/user/simulation_ws/build/catkin_generated/metapackages/fetch_simulation/CMakeLists.txt
-- ~~  - fetchit_challenge
-- ~~  - fetch_gazebo
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)`

Any ideas why this is happening now and how to solve it?
When compiling with the `catkin_make-isolated` compiles but when launching gives:

`Failed to load plugin libfetch_gazebo_plugin.so: libfetch_gazebo_plugin.so: cannot open shared object file: No such file or directory`

RDaneelOlivav, 2019-04-04:
Edit: If you change the Cmakelists.txt to the previous cmake version, compiles, no problem:

cmake_minimum_required(VERSION 3.7.2) --> cmake_minimum_required(VERSION 2.8.3)

Any idea why this could happen?

moriarty, 2019-04-04:
Interesting, I’ll look into it.

moriarty, 2019-04-04:
Confirmed, I can reproduce this.

This doesn't occur with `catkin build` nor `catkin_make_isolated` - which are the only catkin build commands I use :disappointed: 

```
root@3e3b10aeb8dc:/ros/melodic/src# wstool info
workspace: /ros/melodic/src

 Localname    S SCM Version (Spec)     UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------    - --- --------------     -----------  ---------------------------
 fetch_ros      git melodic-devel  (-) 25009ccea0d9 github.com/fetchrobotics/fetch_ros.git
 fetch_gazebo   git gazebo9  (-)       08d39c634818 github.com/fetchrobotics/fetch_gazebo.git
```

```
root@3e3b10aeb8dc:/ros/melodic# catkin_make 
Base path: /ros/melodic
Source space: /ros/melodic/src
Build space: /ros/melodic/build
Devel space: /ros/melodic/devel
Install space: /ros/melodic/install
Creating symlink "/ros/melodic/src/CMakeLists.txt" pointing to "/opt/ros/melodic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /ros/melodic/src -DCATKIN_DEVEL_PREFIX=/ros/melodic/devel -DCMAKE_INSTALL_PREFIX=/ros/melodic/install -G Unix Makefiles" in "/ros/melodic/build"
####
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /ros/melodic/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/melodic
-- This workspace overlays: /opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.15", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /ros/melodic/build/test_results
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.15") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.17
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 13 packages in topological order:
-- ~~  - fetch_calibration
-- ~~  - fetch_description
-- ~~  - fetch_gazebo_demo
-- ~~  - fetch_maps
-- ~~  - fetch_ros (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'fetch_ros' contains non standard content. Use the content of the following file instead: /ros/melodic/build/catkin_generated/metapackages/fetch_ros/CMakeLists.txt
-- ~~  - fetch_simulation (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'fetch_simulation' contains non standard content. Use the content of the following file instead: /ros/melodic/build/catkin_generated/metapackages/fetch_simulation/CMakeLists.txt
-- ~~  - fetchit_challenge
-- ~~  - fetch_navigation
-- ~~  - fetch_teleop
-- ~~  - fetch_gazebo
-- ~~  - fetch_ikfast_plugin
-- ~~  - fetch_depth_layer
-- ~~  - fetch_moveit_config
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:67 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/ros/melodic/build/CMakeFiles/CMakeOutput.log".
See also "/ros/melodic/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

moriarty, 2019-04-04:
My bad... I'll revert the **cmake_minimum_required** I only changed it because bumping it removed a warning from one package, and then I decided to make them all the same.

```CMake
cmake_minimum_required(VERSION 2.8.3)
```

The problem is that **catkin_make** does a direct string compare for the meta-package and only supports **2.8.3** I only changed all the packages for the sake of making them the same.

```CMake
        if(EXISTS ${CMAKE_SOURCE_DIR}/${path}/CMakeLists.txt)
          # compare CMakeLists.txt with standard content
          file(STRINGS ${CMAKE_CURRENT_BINARY_DIR}/catkin_generated/metapackages/${name}/CMakeLists.txt generated_cmakelists)
          file(STRINGS ${path}/CMakeLists.txt existing_cmakelists)
          if(NOT "${generated_cmakelists}" STREQUAL "${existing_cmakelists}")
            set(CATKIN_NONHOMOGENEOUS_WORKSPACE TRUE)
            message("WARNING: The CMakeLists.txt of the metapackage '${name}' contains non standard content. Use the content of the following file instead: ${CMAKE_CURRENT_BINARY_DIR}/catkin_generated/metapackages/${name}/CMakeLists.txt")
          endif()
        else()
          message("WARNING: The metapackage '${name}' has no CMakeLists.txt. Please add one to the package source. You can use the following file: ${CMAKE_CURRENT_BINARY_DIR}/catkin_generated/metapackages/${name}/CMakeLists.txt")
        endif()
```

PR #70 (2019-04-04): catkin_make only supports cmake 2.8.3 for meta-pkg
This fixes #69
commit c663a823 (2019-04-04): catkin_make only supports cmake 2.8.3 for meta-pkg (#70)

This fixes #69
closed 2019-04-04 by moriarty