fetchrobotics/fetch_gazebo #73: [FetchIt!] update tables and walls

moriarty, 2019-04-05:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

RDaneelOlivav, 2019-04-05:
OK I have several questions:
1) The tables and their placement: The Table arrangement, location and dimensions are the same as the one we have already in the simulation? The closer they are to the real world the better. The table covers with the logos, I can start working on them. For the Fetch logo could you send which design you want? The tables flat surface will be the WooodDM material that we see or will it have some panel over it? Because I see that the vertical black metal plates protrude around 2cm from the surface, so that could have a huge impact in grasping. 

2) The Walls. In the simulation, only two walls have been placed. So you say that they will be as the ones in Robocup@home competition? What is their height? Will it be even around the entire perimeter or will it be higher on some sides and lower on the others? I suppose the formate you say is this, without logos I suppose or will it have logos of some kind? I'm being really detailed specific because these things can have a huge impact in perception. We want users that have developed their stuff in the ROSject simulations to be able to make the transition to reality as easy as possible.

[image]

3) I answered in the different issues but, I suppose that the Kit trays we need at least Green, Grey, Yellow, Orange and blue. Also the assembly parts colors, which will they be? Or at least the color assortment so in the simulation, we can even spawn randomly colors objects to train the participants that it could be that way.

4) Also, the lighting conditions are vital. Some time ago we created a ROSject emulating Fetch Domain Randomization Deep Learning paper, we would implement that in the simulation to change slightly the lighting conditions. Let me know what you think: [DomainRandomization](https://www.youtube.com/watch?v=oXsXIia26do&t=242s)

moriarty, 2019-04-05:
I’ll answer in more details when I get to the office:

The tables will be the height we originally sent you, the black bars extend in the photo but I will get white tops for them and the original dimension included the tops.

The side panels with the logos I don’t know the thinkness of, but it might be safe to add 1cm to the dimensions unless we did that originally @velveteenrobot @narora1 ?

The walls will be around most of the arena, with just an opening to get a robot in and out. They are the ones in the photo, and they will only be the short ones- 40 inches tall.

moriarty, 2019-04-05:
> 1. The tables and their placement: The Table arrangement, location and dimensions are the same as the one we have already in the simulation? The closer they are to the real world the better. The [...]

Yes, the original dimensions I sent you was should be just slightly higher than the vertical black posts- I will get tops for the tables this weekend and the tops will be white. When we made the original dimensions we took this into account. The concrete floor where the tables are isn't perfect, so some might be a little tilted but we expect the teams are going to use some sort of **PCL**, **Open3D** or **OpenCV** based detection and segmentation to account for small offsets in the table height.

> 2. The Walls. In the simulation, only two walls have been placed. So you say that they will be as the ones in Robocup@home competition? What is their height? Will it be even around the entire [...]

Yes those are the walls, they won't have the logos on the walls (because they're now being provided by and returned to the venue) The logos will be on the front of all the tables. The walls will be all half height - 40 inches tall - like the one in the photo, we won't use the tall walls. The audience will be able to stand around 3 of the sides and should have a clear view of the competition and also see all the logos.

I'm assuming, that teams won't be using visual slam or visual odometry for navigation because of the location of the camera and laser scanner.

> 3. I answered in the different issues but, I suppose that the Kit trays we need at least Green, Grey, Yellow, Orange and blue. Also the assembly parts colors, which will they be? Or at least the [...]

We have Green, Yellow, Blue, Grey, Orange. We ordered the same kit from several different online shops just to see how much they vary in shape and texture. We wanted to verify that if teams go onto an online shop and order a kit it would be close enough to the real thing.

This is related to #47 it would be best to add all colours, we @fetchrobotics/opensource have not had any follow up questions or concerns related to the colour of the kits, but if teams want to vote for one colour then they should have access to all of the colours in the simulator and then we can discuss it later.

> 4. Also, the lighting conditions are vital. Some time ago we created a ROSject emulating Fetch Domain Randomization Deep Learning paper, we would implement that in the simulation to change slightly [...]

Yes, our location in our office was chosen because we can vary the lighting greatly, it's always bright but we have 2 rows of overhead lights and the two walls to the room a complete glass, with white shutters which we can raise/lower. The time of day and if the teams ask us to raise or lower the shutters will make a difference. If it can be varied in the simulation for them to test slightly various conditions that would be great.

We don't know how the lighting will be in Montreal. The location will be in the ground level of the Convention Centre next to the "Lipstick Forest"  so there might also be some natural light.

https://artpublicmontreal.ca/en/oeuvre/lipstick-forest/

At RoboCup 2018 we had the RoboCup@Home Restaurant Test just a little further down the same hallway. 

> 5. Also reiterate the need to define what exactly does the SHUNK machine do, how it behaves, appearance and so on. I see a @velveteenrobot has created an action client for communicating with the real Shunk but we need to define it in the simulation so people can use it in the same way that in the real place will be used: #72

It simply opens and closes the Chuck:
https://schunk.com/us_en/clamping-technology/series/rota-tps/

The teams need to place the Large Gear into the chuck and call the new action from @velveteenrobot which will close/open the chuck.

RDaneelOlivav, 2019-04-08:
Ok Great answer! Then I'll start working on all this from Wednesday onward, due to other tasks I have. I will start with the environment, tables, and color of elements. Then I'll work on the random lights and other improvements.

moriarty, 2019-04-08:
[image]

Sorry I meant to get you the dimensions on Friday, I wrote them down and didn't get a chance to type them up- so I'll just send an image.

The units are centimeters. T1, T2, T3, T4 are the tables which were measured before so I didn't measure them again.

The opening is currently 76cm. The outside is actually 305 x 305 which is 10ft x 10ft, which isn't want the rulebook said- we used meters but the event in Montreal also wanted the arena specified in ft (even though Canada uses metric- it's very mixed because a lot of building supplies are imported or labeled for the US market).

RDaneelOlivav, 2019-04-08:
Ok , in centimeters is perfect ;)

moriarty, 2019-04-10:
The signs from SICK & E and M have arrived... I've just put them in place, I'll attach them with Velcro this afternoon.

[image]
[image]

RDaneelOlivav, 2019-04-10:
Great! We will try to send you TheConstruct Image at least. Hope tomorrow have some of the changes for the simulation ;)

RDaneelOlivav, 2019-04-12:
@moriarty here you have the table image 
[Uploading Mesa Fetch 92x78,5 cms.pdf…](Table Image)

RDaneelOlivav, 2019-04-12:
I'm working today in the environment disposition, walls and tables

RDaneelOlivav, 2019-04-12:
@moriarty Just created a pull request with the latest changes regarding this matter: https://github.com/fetchrobotics/fetch_gazebo/pull/78

moriarty, 2019-04-16:
FYI @frank-qcd-qk

ben-fido, 2019-04-27:
It looks like some of the tables are wider than others. The dimensions in the Gazebo simulation world need to be checked.

Can someone at Fetch measure the width, depth and height of each table surface and tell us the dimensions?
And the height of the sponsor board? (it's higher than the surface?)

| Table No.     | Description                   | Width | Depth | Height | Sponsor | Height of board |
| --------------- | ----------------------------- | -------- | ------- | -------- | ----------- | ------------------- |
| T1                | Screws in green tub     | ? | ? | ? | E and M | ? |
| T2                | Kits                               | ? | ? | ? | Sick | ? |
| T3                | Gearbox parts on table | ? | ? | ? | E and M | ? |
| T4                | Place completed kits     | ? | ? | ? | Sick | ? |
| T5 (Schunk) | Schunk Machine           | ? | ? | ? | Schunk | ? |

.

| Table No. | Photo                                                                                |
| ------------ | ----------------------------------------------------------------------- | 
| T1            | [image] |
| T2            | [image] |
| T3            | [image] |
| T4            | [image] |
| T5            | [image] |

All dimensions in meters (SI Units).

.

ben-fido, 2019-04-29:
Thanks @velveteenrobot for check the measurements

moriarty, 2019-04-29:
We’re expecting that teams will be able to tolerate some small variance in the real and simulated environment.

The real tables in our environment are missing a 3/4 inch top shelf.

moriarty, 2019-05-12:
Closing this issue as the launch files with “Montreal” in the title are close enough.

moriarty, 2019-05-12:
We don’t have the artwork for the Schunk Machine.

The logos on the tables will be slightly different in the real vs simulated environment.

closed 2019-05-12 by moriarty

moriarty, 2019-05-19:
FYI- the environment in Montreal will change. But you will have time to update your maps.

moriarty, 2019-05-20:
@umhan35 - sorry I’m replying to your email here...

Please note the height in the simulation environment was updated when we knew the height of the walls in Montreal, and we never specified the hight of our walls in our remote access environment- they just happened to be made out of material we had on hand, but the simulation environment has the walls at the height which we specified as soon as we knew what it would be with the local on site contractor- and they can not be changed.

The only thing that has changed, is the location of the tables. Because of the material which the walls are built, and the location of the door- we moved the tables slightly. But there will be enough time tomorrow for everyone to build and test two maps: one per arena, and update some pre configured poses.

umhan35, 2019-05-20:
The walls shouldn’t be a problem for us.