fetchrobotics/fetch_gazebo #74: [FetchIt!] update the colours of the parts

moriarty, 2019-04-05:
**Is your feature request related to a problem? Please describe.**
Currently the large and small gear are green and red.

**Describe the solution you'd like**
They should be grey or blue, or have the option to test with both grey and blue.
Because that's the colour of the 3D print material we have on hand.

**Describe alternatives you've considered**
We don't mind the option for blue and grey. It would give us the flexibility in the future to say the kit needs to be assembled using all matching colours, but for this year we're most likely going to use all one colour.

**Additional context**
I'll add a picture of the parts we've printed so far

moriarty, 2019-04-05:
[image]
[image]

RDaneelOlivav, 2019-04-08:
Ok , understood, I'll start ASAP to implement the changes. If I can I'll create both colors (Blue and Grey ).
Could you send a picture of the blue parts?
You say that in the test they will be all in the same color or in the same test the color might change?

RDaneelOlivav, 2019-04-12:
@moriarty created https://github.com/fetchrobotics/fetch_gazebo/pull/79 to solve this issue.

ben-fido, 2019-04-27:
This ticket can be closed.

jingxixu, 2019-05-02:
Do you mean they can be either blue or grey in the arena? But it seems that you do not provide a blue version of these objects in real world. Many thanks!

moriarty, 2019-06-05:
@jingxixu we originally printed some parts in Blue, and then printed the rest in Grey. All of the blue parts were the original prototypes. We made some modifications to them before deciding on the official parts, which is why the teams didn't have them to test with in the remote sessions.

It takes a lot of time to print the parts, so we asked everyone in the remote sessions if they were okay with just using Grey.

We'd like to leave the Blue parts in the simulation, to allow us to make slight modifications to the competition in the future...

For example, kits must be made from all the same colour, or one color kit must be delivered before another colour kit... which could increase the difficulty slightly but not require rewriting everything.

I'll close this issue now.

closed 2019-06-05 by moriarty