fetchrobotics/fetch_gazebo #77: How to get Fetch and Gazebo9 installed and tested?

ajaytanwani, 2019-04-12:
Hi Fetch Support,

We have an Ubuntu 18.04 machine with Ros Melodic and Gazebo9 installed:

```
adiganapathi@hermes1-ubuntu18:~/catkin_ws$ gazebo --version
Gazebo multi-robot simulator, version 9.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 9.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

adiganapathi@hermes1-ubuntu18:~/catkin_ws$ 
```

We are trying to get the Fetch robot started in the Gazebo simulator. We have read the other issues here and are aware that there are several problems with the simulator, but we would like to even get the simulator *started* so that we can observe these problems. The question is, how to install?

In our Ubuntu 14.04 machines with Fetch installed, running the Fetch on the Gazebo is easy because it just involves installing a package [as shown in your README](http://docs.fetchrobotics.com/gazebo.html).

However, such a package does not appear to be available for Ubuntu 18.04,[ as suggested by this question on Ros Answers](https://answers.ros.org/question/305685/can-robot-models-for-simulations-be-used-in-all-ros-versions/). That question was answered on October 2018, so perhaps there has been an update since then? But we cannot do a sudo apt-get installed. This is what we see after doing a `sudo apt-get update` beforehand:

```
$ sudo apt-get install ros-melodic-fetch-gazebo-demo
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-melodic-fetch-gazebo-demo
```

Next, we figured we would try a "manual" installation. To our knowledge, this is done by cloning this repository into `catkin_ws/src` and then doing `catkin_make` to build. Here is our output:

```
adiganapathi@hermes1-ubuntu18:~$ mkdir -p ~/catkin_ws/src
adiganapathi@hermes1-ubuntu18:~$ cd catkin_ws/src/
adiganapathi@hermes1-ubuntu18:~/catkin_ws/src$ git clone https://github.com/fetchrobotics/fetch_gazebo.git
Cloning into 'fetch_gazebo'...
remote: Enumerating objects: 34, done.
remote: Counting objects: 100% (34/34), done.
remote: Compressing objects: 100% (19/19), done.
remote: Total 1167 (delta 14), reused 33 (delta 14), pack-reused 1133
Receiving objects: 100% (1167/1167), 16.86 MiB | 31.00 MiB/s, done.
Resolving deltas: 100% (606/606), done.
adiganapathi@hermes1-ubuntu18:~/catkin_ws/src$ cd ..
```

And then running `catkin_make` results in the following [output in the following pastebin](https://pastebin.com/GVv9scKT).

It suggests to install some "robot_controllers" package, but we are not sure what that refers to, or if installing this is on the right track towards getting the Fetch displayed in the Gazebo.

Hopefully this issue shows what we are trying to do --- just getting the Fetch displayed in our Gazebo. Are these the right steps to follow? If not, where may we find the steps?

We're happy to provide any more information about our setup if necessary.

Thanks for your time!

moriarty, 2019-04-12:
Hi,

I’ll need to check if it’s available in the standard ROS repos yet or if it’s only in shadow-fixed...

https://github.com/ros/rosdistro/pull/20745 Was merged 13 days ago, and if there hasn’t been a “Melodic sync” ( a process manually triggered by someone at Open Robotics / OSRF ) then it will only be in the shadow-fixed repository

moriarty, 2019-04-12:
But, when you’ve installed from source, you’ll need to run a rosdep command to install the dependencies you need.

You can see an example in this dockerfile I shared:

https://github.com/fetchrobotics/fetch_gazebo/issues/75

ajaytanwani, 2019-04-12:
rosdep install fixed it.

moriarty, 2019-04-12:
I also responded to @DanielTakeshi in https://github.com/ros/rosdistro/issues/20745

closed 2019-04-12 by moriarty

DanielTakeshi, 2019-04-12:
Thanks for the clarifications @moriarty . It's really nice that you are responsive.

RDaneelOlivav, 2019-04-12:
@ajaytanwani You can also use the ROSject with the simulation prepared there: [ROSJECT.Fetchit-Challenge](http://www.rosject.io/l/8361c2a/)
You can pull the gits to get the latest changes.