fetchrobotics/fetch_gazebo #81: [FetchIt!] Update the screw bin in simulation

umhan35, 2019-04-23:
@moriarty We saw last week the bin containing screws is updated. Could you update the simulation? @RDaneelOlivav Having the CAD file should be good for now, so we can add ourselves.

Sarah gives the samzon link: https://www.amazon.com/dp/B00K2NZSFM/

[image]

RDaneelOlivav, 2019-04-23:
I wasnt aware of that. You mean that the Bin where the screws are placed has been changed from the small ones to this big bucket like box? @moriarty ?
I'll put in the issues line to change it ;). But I would appreciate more info about this to make it right.

moriarty, 2019-04-23:
@RDaneelOlivav yes sorry we didn't update you while you were on holiday, and we were still makine up our minds.

@velveteenrobot ordered new bins after the first teams accessed the remote area and requested more clearance for the wrist.

We went online and ordered a few difference options and we have been testing them and have now decided on this version.

This is the bin:

https://www.amazon.com/dp/B00K2NZSFM

RDaneelOlivav, 2019-04-23:
Great! ;). So both the screews and the upper and lower assembly parts will be in these containers? Any colour in particular? All of them are green?

ben-fido, 2019-04-28:
The two halves of the gearbox will be placed directly on the table surface. No bin.

The screws will be in this green tub:  
https://www.amazon.com/dp/B00K2NZSFM/

 [image]  [image]

.

The dimensions are (in meters):

|                        |  Width on bottom surface | Width at top | Depth on bottom surface | Depth at top | Height |
| ----------------- | -------------------------------- | ---------------- | ------- | -- | -- |
| Inside the tub | 0.450  | 0.470  | 0.335 | 0.350 | 0.125  |
| Outside          |  0.455 |  0.475 | 0.340  | 0.355 |  0.130  |

.

Those dimensions don't include the curved lip around the edges or the handles.

The lip sticks-out 0.020m (all the way around) and is 0.010m high.

The handles are 0.100m along the depth axis, they stick-out 0.040m along the width axis, and are 0.025m high.

.

Is someone going to make a 3D model?

.

RDaneelOlivav, 2019-04-29:
Thats one of my tasks this week if all goes well ;)

moriarty, 2019-04-29:
Info from the bottom of the bin:

> CARLISLE
> OKC, OK
www.carlislefsp.com
> N44010

https://www.carlislefsp.com/warewashing/tote-box/N4401009

moriarty, 2019-06-05:
Can we close this one too?

umhan35, 2019-06-05:
Can we add the green bin to Gazebo?

closed 2020-08-09 by erelson