fetchrobotics/fetch_gazebo #82: [FetchIt!] SCHUNK machine model shoudn't have the cylinder inserted

umhan35, 2019-04-23:
@moriarty Our team just know that we need to insert the gear inside the whole of the SCHUNK machine. In simulation, there is a half yellow half black cylinder inserted and should be removed.

@RDaneelOlivav  We tried to use Blender to remove the cylindar from `shunk_machine_lathe_centered.dae` but the main.launch file won't launch probably because the pose also needs to be updated?

In sim:
[image]

In the arena:
[image]

moriarty, 2019-04-23:
@ajzeller Could you open up the model in Solid Works and remove the object from the lathe?

Then we can give a new model to @RDaneelOlivav , unless @RDaneelOlivav is able to do this directly.

ajzeller, 2019-04-23:
@moriarty here is the URDF with the object removed:

[schunk_machine_URDF_4-23-19.zip](https://github.com/fetchrobotics/fetch_gazebo/files/3109649/schunk_machine_URDF_4-23-19.zip)

[image]

moriarty, 2019-04-23:
Thanks @ajzeller !

Actually could you add one more thing?

The real machine has boards around it, then we can fix #61 and this issue at the same time.
Don't worry about the logo, just the boards for now as they should be their for the teams to make and test their mapping and localization.

ajzeller, 2019-04-23:
@moriarty yep here you go:

[schunk_machine_URDF_4-23-19_v2.zip](https://github.com/fetchrobotics/fetch_gazebo/files/3109949/schunk_machine_URDF_4-23-19_v2.zip)

RDaneelOlivav, 2019-04-24:
Ok great! I'll put it in the task list to do ;)

moriarty, 2019-04-24:
This photo was requested by @banerjs 
[image]

moriarty, 2019-04-25:
@RDaneelOlivav just as a clarification re:

> Also I haven't integrated yet the SCHUNK actions into the machine, but I think that's for another time.

The action **_only_** opens and closes the Schunk **chuck** not the whole door. The Door, is on sliders, which the teams can push on the handle to slide open and closed.

ben-fido, 2019-04-28:
The above colored images of the Schunk Machine are from a CAD program.

We downloaded the .zip file and opened the 3D model.  
It looks like this:  
[image]
It is one single solid part.  
This would be a regression in functionality, because @RDaneelOlivav already did a great job to build the current model:  It has multiple parts, color texture, the door slides shut and the spindle can rotate by publishing a ROS msg command.

.

Have a look at this Collada file:  
`fetch_gazebo/fetchit_challenge/models/shunk_machine/meshes/shunk_machine_lathe_centered.dae`
[image]

We just need someone to open this model and bore out the tube.

Still, the chuck jaws do not open and close. I will make a new ticket for that.

.

RDaneelOlivav, 2019-05-03:
Ok I've pushed an update of the chunk machine ( #89), with new updated action server decoupled from simulation or real thing. Also updated the model, colours, and added chuck open and close system, with integration with the action server. Now if you spawn the Shunk machine it should just be there operational.
To use it just publish the goal in the goal topic.

moriarty, 2019-06-05:
Fixed with #89

closed 2019-06-05 by moriarty