fetchrobotics/fetch_gazebo #83: [FetchIt!] [BUG] Gazebo world missing romanoff utility caddy

ben-fido, 2019-04-24:
**Description:**
The kits (Romanoff utility caddy) are missing from some of the Gazebo worlds in the `fetchit_challenge` package.  I think the model URL is wrong and it's trying to download from the internet.

**Steps to reproduce:**
Launch any one of these 4 launch files. They all have the same problem:  
```
roslaunch fetchit_challenge fetchit_challenge_arena_montreal2019_highlights.launch
roslaunch fetchit_challenge fetchit_challenge_arena_montreal2019.launch
roslaunch fetchit_challenge main_arena_montreal2019_highlights.launch
roslaunch fetchit_challenge main_arena_montreal2019.launch
```
In the terminal you see this error message:   
```
[Err] [ModelDatabase.cc:414] Unable to download model[model://romanoff_small_utility_caddy]
```

Here is the full output:   
```
[ INFO] [1556064604.854347292]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1556064604.855203264]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.0.6
[ INFO] [1556064604.898350072]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1556064604.899131424]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.0.6
[Wrn] [ModelDatabase.cc:340] Getting models from[http://gazebosim.org/models/]. This may take a few seconds.
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[Err] [ModelDatabase.cc:414] Unable to download model[model://romanoff_small_utility_caddy]
[Err] [SystemPaths.cc:412] File or path does not exist[""]
Error Code: 11 Msg: Unable to find uri[model://romanoff_small_utility_caddy]
[ INFO] [1556064624.105415954, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1556064624.123060014, 0.039000000]: Physics dynamic reconfigure ready.

```

**Screenshot:**
Here is a screenshot from "main arena montreal 2019", on the left you can see empty table.

[image]

**catkin workspace:**
 - fetch_gazebo version: Current master branch
 - Ubuntu version: 18.04
 - ROS version: Melodic

.

RDaneelOlivav, 2019-04-24:
I'll have a look at this

RDaneelOlivav, 2019-04-24:
Fixed in: https://github.com/fetchrobotics/fetch_gazebo/pull/85
As soon as it's merged into the main repo gazebo9 branch you should be able to see the cart in all the worlds. I've also updated the assembly elements colors so that they are the grey ones that will be used in the final competition.

RDaneelOlivav, 2019-04-24:
Oh I didnt see that you need 3 kits on the table. I'm adding right now three of different colour.
Let me know if its ok

RDaneelOlivav, 2019-04-24:
Done. Now 3 carts appear green, yellow and red. You can change them in the world for any of the colors you want to test your algorithms are robust.
[image]

ben-fido, 2019-04-24:
Thanks @RDaneelOlivav

closed 2019-04-25 by ben-fido