fetchrobotics/fetch_gazebo #84: [FetchIt!] Final list of objects and simulation layout

ben-fido, 2019-04-24:
**Issue description:**
The object types have not been finalized (https://github.com/fetchrobotics/fetch_gazebo/issues/81) and the simulation world is out of date.

**List of objects:**
Can Fetch Robotics provide the final information for these:
1. Kit
 - Exact model from Amazon  
Maybe Romanoff or something else.

2. Bin containing screws.
 - Exact model from Amazon.
Maybe https://www.amazon.com/dp/B00K2NZSFM/
 - LxWxH measurements.

3. Bin containing gearbox parts.
 - Exact model from Amazon.
 - LxWxH measurements.

It seems fair to use the exact object models for an industrial robot, and we want to make new 3D models.

**Fixes to simulation:**
Can someone at Fetch Robotics edit the simulation world?  
 - Missing wall on front of Schunk table.  (https://github.com/fetchrobotics/fetch_gazebo/issues/82)
 - There's a part in the chuck of the Schunk.  (https://github.com/fetchrobotics/fetch_gazebo/issues/82)
 - No kits are spawned.  (https://github.com/fetchrobotics/fetch_gazebo/issues/83)
 - Wrong type of bin containing screws.
 - Wrong type of bin containing gearbox parts.

.

RDaneelOlivav, 2019-04-24:
Let me check that as soon as possible.

RDaneelOlivav, 2019-04-24:
In theory, solved the issue of the kits not appearing. Pending of merge : https://github.com/fetchrobotics/fetch_gazebo/pull/85

Next issue to tackle will be the Shunk model update, maybe the action integration.

moriarty, 2019-04-24:
@RDaneelOlivav & @ben-fido Thanks for bringing this up.

> 1. Kit
https://www.amazon.com/gp/product/B074Q2Y7WR

> 2. Bin containing screws.
https://www.amazon.com/dp/B00K2NZSFM

> 3. Bin containing gearbox parts.

Actually the 3rd Bin fell off of my radar and we forgot to buy it.

The original plan, was to have a "bin" which was actually a cardboard box that contained some foam which would evenly space the parts... This is how some parts are actually shipped in industry and it would be as if the Fetch was just taking the parts out of open boxes.

FYI @velveteenrobot & @cjds  & @narora1

ben-fido, 2019-04-25:
Can you please decide which bin?

moriarty, 2019-04-25:
I’ll let you know by the end of the week. But for now @RDaneelOlivav perhaps just put the gearbox top and gearbox bottom on the shelf without a bin.

moriarty, 2019-06-05:
We opted not to have the 3rd Bin because there wasn't enough time to get one and notify the teams.

So items 1 & 2 were used and 3 was dropped.

closed 2019-06-05 by moriarty