fetchrobotics/fetch_gazebo #87: [FetchIt!] Modify chuck on Schunk machine

ben-fido, 2019-04-28:
**Background:**
The real Schunk machine has an actuated chuck with jaws that can be opened and closed.  
There is a ROS Node running on the network with a SimpleActionServer. You can publish a Goal msg to open and close the chuck.

[image]

This video shows you the pneumatic chuck with the moving jaws:  
https://www.youtube.com/watch?v=Cl5XYM2xK1E
When the jaws are open, you can slide in your part.  
Then you close the jaws and part stays there.

This functionality needs to be added to the simulation.

**Tasks required:**

 - Which model number is the chuck?
https://schunk.com/us_en/clamping-technology/series/rota-tp/  
Then we can download the CAD model directly from Schunk.  

 - What is that black part on the chuck?
We can't find it on the above website.

 - Make a new Collada 3D model of spindle, chuck and jaws with separate parts.
It should be similar to this model:   [image]  
`fetch_gazebo/fetchit_challenge/models/shunk_machine/meshes/shunk_machine_lathe_centered.dae`  
but with the 3 jaws removed.

 - Modify URDF of Schunk Machine.  
There is one link for the body of the machine.
There is another link for the spindle + chuck.
And 3 links for the 3 jaws, each rotated by 120 degrees. The Collada model for one jaw can be used 3 times.  

 - Make a simple joint controller.  
We can move the 3 joints together using a simple controller,  
or maybe hack some "mimic joint" like this:  
https://answers.ros.org/question/283537/how-to-do-mimic-joints-that-work-in-gazebo/

 - Modify code in the SimpleActionServer:  
`fetch_gazebo/fetchit_challenge/nodes$ pico schunk_machine_server.py`  
If you launch it with the `sim` flag, it will relay commands to the Gazebo controller.  
Otherwise it sends command to the GPIO port (the real hardware).  

.

RDaneelOlivav, 2019-04-29:
We actually have in the issues queue to update the SHUNK model to the latest configuration. This of course should be a nice adition to the simulation for sure, at least as far as Gazebo simulation can take it.

ben-fido, 2019-04-29:
**Which model number is the shunk power chuck?**
If you don't know, can you send us the CAD files you receive from shunk?
@moriarty @ajzeller

moriarty, 2019-04-30:
Sorry, we’re able to release the models but not the cad files directly

ben-fido, 2019-05-06:
@RDaneelOlivav the new changes to machine in PR #89 are very good
We tested on kinetic and melodic.

Open / close
[image]

New joint axes
[image]

If the axes of machine door is moved closer to the object itself, it might slide more easy
[image]

.

moriarty, 2019-05-08:
@ben-fido I've just merged #89 can this now be closed?

closed 2019-06-05 by moriarty