fetchrobotics/fetch_gazebo #90: [Fetchit!] [BUG] Rotated gears "stand up" over time

GregoryLeMasurier, 2019-05-07:
@RDaneelOlivav When I rotate or spawn in a gear on its side it will start to stand up as seen in the pictures below.
Being able to rotate a gear in simulation would be really useful for testing because in the competition some of the gears will be in this orientation.

**To Reproduce**
Change the pose roll, pitch, or yaw of any gear to be 1.2.
Change the z of the gear to be above the schunk table (0.83)

Gear after rotation is applied:
[image]

Gear a few seconds after rotation is applied:
[image]

RDaneelOlivav, 2019-05-09:
@moriarty @GregoryLeMasurier I'll solve it tomorrow, is that ok?

moriarty, 2019-05-09:
Yes Tomorrow should b fine. @GregoryLeMasurier I merged in some changes yesterday which I think should have updated the appearance of the Schunk machine.

RDaneelOlivav, 2019-05-10:
Ok , I've made a pull request with the changes in the inertias needed to solve this issue: https://github.com/fetchrobotics/fetch_gazebo/pull/92

As soon as this is merged I'll post a ROSJect with the latest changes ;) for you people to use.

closed 2019-05-10 by moriarty