fetchrobotics/fetch_gazebo #91: fetch robot is dark in gazebo simulation playground

jingxixu, 2019-05-10:
**Describe the bug**
Fetch robot is dark in the gazebo simulated playground.

**To Reproduce**
ROS Melodic, Ubuntu 18.04, Gazebo9
```
roslaunch fetch_gazebo playground.launch
```

**Expected behavior**
robot should not be dark

**Screenshots**
[image]

**catkin workspace (please complete the following information):**
 - fetch_gazebo version: Gazebo 9
 - Ubuntu version: 18.04
 - ROS version: Melodic
 - current catkin workspace information `wstool info`

**Additional context**
None

moriarty, 2019-05-10:
This doesn’t effect the simulation.

This is just Gazebo visualization.

The work around- use a GPU with the drivers installed properly.

Or ignore it.

I have a Dell XPS 13 without a GPU, and gazebo visually appears dark.
But on my desktop with a GPU, it’s fine.

The data sent to the simulated sensors shouldn’t be effected by this.

moriarty, 2019-05-10:
@RDaneelOlivav just wanted a second opinion on this before I close it as “won’t fix” because as far as I know this is just the rendering of the robot, and doesn’t effect the simulation.

RDaneelOlivav, 2019-05-10:
Absolutely agree with you @moriarty . Sure the graphics can always be tweaked in the simulation to get better but the important part is what the cameras see, the GUI side is just for us humans to understand what's going on. Your robot only cares about the sensor side.

closed 2019-05-10 by moriarty