fetchrobotics/fetch_gazebo #94: [BUG] Arm gets into unusable position when using moveit and moving the torso

GregoryLeMasurier, 2019-06-12:
**Describe the bug**
The arm gets into an unusable state after controlling it with moveit and then moving the torso.
The arm seems to fall down and is able to go through the other parts of the robot. We can not control the arm after the arm gets into this state.

**To Reproduce**
Clone this repo: (https://github.com/GregoryLeMasurier/FetchArmTest)

Launch gazebo with fetch:
roslaunch fetchit_challenge main_arena_montreal2019_highlights.launch

Start move group:
roslaunch fetch_moveit_config move_group.launch

Move the arm:
rosrun test_ws arm_out_node

Move the torso:
rosrun test_ws torso_up_node

Try to move the arm while it is stuck in robot:
rosrun test_ws arm_out_node

**Expected behavior**
Arm doesn't move or fall after the torso moves. 

**Video**
(https://youtu.be/yihiZyimFCI)

**catkin workspace (please complete the following information):**
 - fetch_gazebo version: gazebo9
 - Ubuntu version: 18.04
 - ROS version: Melodic

umhan35, 2019-06-12:
@RDaneelOlivav this is the issue that I mentioned when you interview us during the competition.

moriarty, 2019-06-12:
Thumbs up :+1: for the great bug report!

I'm currently working remotely from my personal laptop, so this may need to wait until next week when I'm back in the office.

RDaneelOlivav, 2019-06-12:
@umhan35 I think you are confusing me with Ricardo , my boss, he was the one going to the competition ;).
Have you checked the status of the controllers?
Does it happen in an empty world or if you only move the arms, without moving the torso?

GregoryLeMasurier, 2019-06-12:
@RDaneelOlivav This happens in an empty world, I used: roslaunch fetch_gazebo simulation.launch. It does not happen when we only move the arm. (I have updated the repo in the first post to include another node for moving the arm). I am not sure how to check the status of the controllers. I saw some rostopics for controllers but I didn't find any data that seemed useful for this issue.

moriarty, 2019-06-12:
I think @RDaneelOlivav is right... there is this file:

@GregoryLeMasurier it looks like the launch file you’re using calls:

https://github.com/fetchrobotics/fetch_gazebo/blob/gazebo9/fetch_gazebo/launch/include/fetch.launch.xml

But you want to call the one with the **_pp**

https://github.com/fetchrobotics/fetch_gazebo/blob/gazebo9/fetch_gazebo/launch/include/fetch_pp.launch.xml

The **_pp** I assume stands for pick & place, as that launch file calls:
prepare_simulated_robot_pick_place.py
... but I’m not sure why this isn’t the default as I’d expect if you’re using the Fetch in simulation you always want to wait for the arm controllers.

moriarty, 2019-06-12:
Maybe ignore my comment- I’m on my phone and can’t really easily view the diff between the scripts.

GregoryLeMasurier, 2019-06-13:
@moriarty  I was originally using main_arena_montreal2019_highlights.launch which does call the pick and place version. I also have tried simulation.launch which does use fetch.launch.xml. Both versions have the same issue.

GregoryLeMasurier, 2019-06-24:
@moriarty Have you had a chance to look into this yet?

moriarty, 2019-07-25:
No sorry - after ICRA I had some catching up to do, was at RoboCup, and missed some days for personal reasons, I'll be away next week too.

I'll assign @erelson

erelson, 2019-07-27:
Sorry, I won't be able to dig into this. My experience level with Gazebo is zero, so this isn't something I can productively troubleshoot.

moriarty, 2019-07-29:
Okay- I’ll look into it after I get back from Vacation.