fetchrobotics/fetch_gazebo #96: [feature] Migrating to ROS2

ruffsl, 2019-10-16:
Our research lab is in the progress of migrating our software stack to ROS2, and we like to start by testing out these infrastructure changes in simulation when doing so. I'd like to ask if there are any ongoing plans or current progress in updating the fetch ecosystem to use ROS2, be it in general for the hardware platform, or just for gazebo simulation, as we'd like to avoid duplicate efforts. We'd like to reach out to not only to Fetch, but any end user/developers to pool together any contributions.

We are also exploring the use of the ros2/ros1_bridge , but to me that's a short term solution.

mkhansenbot, 2021-02-09:
What ever happened to this effort? Did you upgrade your fetch to ROS2?

ruffsl, 2021-02-09:
I tried to dive into migrating the gazebo simulation and controllers about the time I made the first post so that we'd have something to debug with that wouldn't hog so much physical lab hardware, but got blocked by the state of ros2_control and gazebo_ros_pkgs. My hunch is that those two packages have greatly matured since then, but I now no longer have the lab time to devote to this, so we've been making due with ros1_bridge. I'm not sure if fetch has open sourced there motor controller nodes, so I'm not sure how much further this could be done without Fetch migrating on their end.

SteveMacenski, 2021-06-18:
Also interested in using this for ROS2 in a book!

ruffsl, 2021-06-18:
@SteveMacenski , @mkhansenbot , I recently revisited this, as I was helping @aefrank with getting at least a simulation of the Fetch research platform working in Gazebo with ROS2 (w/ or w/o bridging):

https://github.com/fetchrobotics/power_msgs/pull/13
https://github.com/fetchrobotics/fetch_msgs/pull/19
https://github.com/CogRob/fetch_ros/pull/3
https://github.com/CogRob/fetch_gazebo/pull/2

But I don't really feel up to committing to a ROS2 port of the original codebase if Fetch Robotics Inc isn't initiating it themselves. As mentioned, the original code base for the fetch gazebo plugin is rather relient on robot_controllers which is specific to Fetch, and I'm not sure who else is familiar with it:

https://github.com/fetchrobotics/robot_controllers

So I was thinking we could try and re-use the same URDFs and model files and make use of the recently developed gazebo_ros2_control package that has more active maintainers:  

https://github.com/ros-simulation/gazebo_ros2_control

The example demos for which seem a lot simpler to write and compose together: 

https://github.com/ros-simulation/gazebo_ros2_control/tree/master/gazebo_ros2_control_demos/examples

mikeferguson, 2021-06-19:
In addition to being the primary author of robot_controllers, I've actually ported robot_controllers to ROS2 and am a maintainer of the ROS2 port - it's fully working on 3 of my personal robots and I see no reason it wouldn't work on the Fetch Simulation: https://github.com/fetchrobotics/robot_controllers/tree/ros2

While I don't have the time to port the Fetch simulation to ROS2, if any issues came up with robot_controllers, I can certainly help with those.

mikeferguson, 2021-06-19:
I should also point out that I forked the depth_layer plugin from fetch_ros and ported it to ros2/navigation2 for my UBR-1 robot: https://github.com/mikeferguson/ubr_reloaded/blob/ros2/ubr1_navigation/src/depth_layer.cpp

aefrank, 2021-06-19:
@mikeferguson This is great to hear, very grateful that someone has already tackled this! I will likely be working on using this with Fetch in simulation in the coming weeks, so I will let you know if I have any issues or questions.

mkhansenbot, 2021-07-01:
@aefrank please update the thread if you get a working ROS 2 simulation of fetch_gazebo. Maybe we can get @velveteenrobot to merge it int this repo for everyone to use.

aefrank, 2021-07-01:
@mkhansenbot Sadly I haven't been able to get much working. After trying to work out all the dependencies that would potentially need to be updated to work with ROS 2, I think this is a bit over my head. I haven't done enough work with ROS 2 and unfortunately I don't have the time to do the amount of digging required. I will certainly post back here if I go back to it or have any flash of insight, but for now I'm just going to be working in ROS 1.

Mechazo11, 2024-03-27:
Hi @ruffsl is the port to ROS2 still ongoing?

ruffsl, 2024-03-27:
@Mechazo11 , sorry, no idea.
I've moved and no longer have access to this platform.
Looks like there's been no upstream activity in 3 years. :/

Mechazo11, 2024-03-27:
@ruffsl thanks for letting me know. Any idea where I can begin or any other group who is working on this presently?

ruffsl, 2024-03-27:
> Any idea where I can begin or any other group who is working on this presently?

You could start by taking a look through fork network activity.
- https://github.com/ZebraDevs/fetch_gazebo/network
- https://github.com/ZebraDevs/fetch_gazebo/forks

I've found code search engines a nice resource in sleuthing for leads and related development:
- https://github.com/search?q=fetch_simulation+language%3AXML&type=code
- https://grep.app/search?q=fetch_simulation

I didn't see any commits related ROS 2, so you may want make a post to ROS Discourse and Gazebosim comunity:
- https://discourse.ros.org/
- https://community.gazebosim.org/

Lastly, you could check the gazebo channel on the OSRF discourd server to find anyone else interested or knowledgeable:
- https://discourse.ros.org/t/announcing-the-official-osrf-discord-server/30236

moriarty, 2024-03-27:
🪦 💀 @erelson or @velveteenrobot any internal efforts on this front? I suspect not

Mechazo11, 2024-03-27:
@ruffsl thank you very much for the detailed response. I will look into the resources as suggested

Mechazo11, 2024-03-28:
@ruffsl Hi, after some changes made to a few files, I was successful in opening the basic simulation, ```roslaunch fetch_gazebo simulation.launch``` to work in ROS Noetic. It appears a number of .py files were calling the python 2 interpretor using ```#!/usr/bin/env python``` shebang but changing them to ```#!/usr/bin/env python3``` solved a number of broken issues. I shall post some more updates as I progress with the more complicated simulations.