fetchrobotics/fetch_gazebo #98: Adding namespace to Fetch Robot in Gazebo

salonijain12, 2019-11-07:
Hi team,

I am trying to create a world in Gazebo comprising of multiple robots and for which I need to add namespace to Fetch robot.
Please help me with the steps/ procedure for same.

Thanks and Regards,
Saloni

salonijain12, 2019-11-13:
Hi Team, 

I have done following changes to add Namespace to fetch robot using following steps:

1) Created an initial launch file:

<launch>
  <env name="GAZEBO_MODEL_PATH" value="$(find fetch_gazebo)/models:$(optenv GAZEBO_MODEL_PATH)" />

  <arg name="debug" default="false"/>
  <arg name="gui" default="true"/>
  <arg name="headless" default="false"/>

  #####  Start Gazebo with a blank world ####
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find fetch_gazebo)/worlds/cube.sdf"/>
    <arg name="debug" value="$(arg debug)" />
    <arg name="gui" value="$(arg gui)" />
    <arg name="paused" value="false"/>
    <arg name="use_sim_time" value="true"/>
    <arg name="headless" value="$(arg headless)"/>
  </include>

  <include file="$(find fetch_gazebo)/launch/test.launch" />
</launch>

2) Added group tag in test.launch file
<launch>
  <arg name="robot" default="fetch"/>
  <param name="robot_description" command="$(find xacro)/xacro.py $(find fetch_gazebo)/robots/fetch.gazebo.xacro"/>

  #### Include Robot ####
  <group ns="fetch">
    <param name="tf_prefix" value="fetch" />
    <include file="$(find fetch_gazebo)/launch/fetch1.launch.xml">
      <arg name="x" value="5"/>
      <arg name="y" value="0.0"/>
      <arg name="z" value="0.0"/>
      <arg name="yaw" value="-3.0"/>
    </include>
  </group>
</launch>

3) fetch1.launch.xml

<launch>
  <arg name="x" default="0.0" />
  <arg name="y" default="0.0" />
  <arg name="z" default="0.0" />
  <arg name="yaw" default="0.0" />
  <arg name="robot_name"/>
  <arg name="ns" default="fetch" />

  <rosparam command="load" file="$(find fetch_gazebo)/config/default_controllers.yaml"/>

  
  <node name="robot_state_publisher_fetch" pkg="robot_state_publisher" type="robot_state_publisher" >
    <param name="publish_frequency" value="100.0"/>
  </node>

  
  <param name="robot_description" command="$(find xacro)/xacro.py $(find fetch_gazebo)/robots/fetch.gazebo.xacro"/>
  <node name="fetch_urdf_spawner" pkg="gazebo_ros" type="spawn_model" respawn="false" output="screen"
        args="-urdf -x $(arg x) -y $(arg y) -z $(arg z) -Y $(arg yaw) -model fetch -param robot_description">
    <remap from="robot_description" to="/fetch/robot_description" />
  </node>

  <node name="prepare_robot" pkg="fetch_gazebo" type="prepare_simulated_robot.py" />

  
[... 284 lines cut ...]
[ERROR] [1573633365.981512611, 0.166000000]: No p gain specified for pid.  Namespace: /gazebo/wrist_flex_joint/position
[ERROR] [1573633365.981857964, 0.166000000]: No p gain specified for pid.  Namespace: /gazebo/wrist_flex_joint/velocity
[ERROR] [1573633365.982223015, 0.166000000]: No p gain specified for pid.  Namespace: /gazebo/wrist_roll_joint/position
[ERROR] [1573633365.982599420, 0.166000000]: No p gain specified for pid.  Namespace: /gazebo/wrist_roll_joint/velocity
[ERROR] [1573633365.983004815, 0.166000000]: No p gain specified for pid.  Namespace: /gazebo/l_gripper_finger_joint/position
[ERROR] [1573633365.983412570, 0.166000000]: No p gain specified for pid.  Namespace: /gazebo/l_gripper_finger_joint/velocity
[ERROR] [1573633365.983796716, 0.166000000]: No p gain specified for pid.  Namespace: /gazebo/r_gripper_finger_joint/position
[ERROR] [1573633365.984230458, 0.166000000]: No p gain specified for pid.  Namespace: /gazebo/r_gripper_finger_joint/velocity
[ WARN] [1573633365.984671619, 0.166000000]: **No controllers loaded.**
[ INFO] [1573633365.985270915, 0.166000000]: Finished initializing FetchGazeboPlugin

I think the namespace issue due to which controllers are not loaded from "libfetch_gazebo_plugin.so" which is called from fetch.gazebo.xacro.

I have also tried to add robotNamespace in xacro:
  
  <gazebo>
    <plugin name="fetch_gazebo_plugin" filename="libfetch_gazebo_plugin.so">
      <robotNamespace>fetch</robotNamespace>
    </plugin>
  </gazebo>

But still facing the same error. Please help to resolve the issue.

Thanks and Regards,
Saloni

gdhucoder, 2020-02-28:
Hey  bro! I have the similar issue. I found this in the closed issues. https://github.com/fetchrobotics/fetch_gazebo/issues/18  . If you have any possible solution, please let me konw.