fetchrobotics/fetch_gazebo #103: how to make fetch  execute a pre-defined trajectory in gazebo.

wangjiwang, 2020-07-02:
I don't have any hardware now, and I download this package. how to make fetch  execute a pre-defined trajectory in gazebo? can anybody help me and tell me more details?

erelson, 2020-07-02:
Hi @wangjiwang, the intro documentation should get you started: http://docs.fetchrobotics.com/gazebo.html

The code behind these examples is in the demos folder of this repo: https://github.com/fetchrobotics/fetch_gazebo/tree/gazebo9/fetch_gazebo_demo/scripts

closed 2020-07-02 by erelson

wangjiwang, 2020-07-05:
Thanks for your reply. And I try to use
arm_with_torso_controller/follow_joint_trajectory controller to execute a
pre-defined trajectory. It did not work. The API did not give detail
information. Can you give a example? The attachment is my code.

wangjiwang, 2020-07-14:
> Hi @wangjiwang, the intro documentation should get you started: http://docs.fetchrobotics.com/gazebo.html
> 
> The code behind these examples is in the demos folder of this repo: https://github.com/fetchrobotics/fetch_gazebo/tree/gazebo9/fetch_gazebo_demo/scripts

Thx, Can you tell me the detail information about how to use arm_with_torso_controller/follow_joint_trajectory controller ?