fetchrobotics/fetch_gazebo #106: [BUG] Fetch Robot doesn't tries to pick up the cube

SarveshRobotics, 2020-08-08:
I am launching the default simulation of picking up a cube from the table, built in fetch_gazebo package. The robot simply goes to the prepare position and doesn't move at all. (Initially I was working on a warehouse project with Fetch robot and was trying to perform SLAM with Laser and Odom but it was not working out, hence I started testing the default launch simulations.)

I installed fetch package using git clone. And I have all the packages installed as shown in the screenshot.
[image]
When I run `roslaunch fetch_gazebo simple_grasp.launch ` , the model succesfully launches and comes to prepare pose and stops there. I get two warnings on the console as shown:
[image]
This is the setup:
[image]

I expect the robot to pick up the object successfully.

catkin workspace:
 - fetch_gazebo version: Gazebo9
 - Ubuntu version: 18.04
 - ROS version:Melodic

cjds, 2020-08-17:
Can confirm that this works on my machine. Something must be different about the setup. 

Is it possible to post the goal and result here please?

mikeferguson, 2020-12-01:
So, "simple_grasp.launch" just sets up the environment. You would also need to "roslaunch fetch_gazebo_demo pick_place_demo.launch" after the environment is fully setup.