fetchrobotics/fetch_gazebo #107: About the demo.py in fetch_gazebo_demo

xiangxiang123456, 2020-09-11:
When the simulation environment got ready, we ran the demo.py. And after the robot arriving the first table, errors occurred. It look likes this

`Traceback (most recent call last):
  File "/home/lwx/fetch_ws/src/fetch_tc/fetch_sim/fetch_gazebo_demo/scripts/demo.py", line 276, in <module>
    grasping_client.updateScene()
  File "/home/lwx/fetch_ws/src/fetch_tc/fetch_sim/fetch_gazebo_demo/scripts/demo.py", line 148, in updateScene
    for obj in find_result.objects:
AttributeError: 'NoneType' object has no attribute 'objects'

It seems that the robot could not find the cube. I sincerely hope someone can help us! Thanks a lot!

erelson, 2020-09-11:
Hi @xiangxiang123456 can you clarify what version of gazebo and what version of ROS you are using?

I'm not familiar with this code, but looks like something unexpected in some manner is being returned [here](https://github.com/fetchrobotics/fetch_gazebo/blob/gazebo9/fetch_gazebo_demo/scripts/demo.py#L135), and the traceback results a few lines below when it's used.

xiangxiang123456, 2020-09-12:
Thank you for your reply. @erelson  We are using ros kinetic and gazebo 7. We also found it's matter with this line of code`find_result = self.find_client.get_result()`.  After printing out the "find_result", It is NONE.